iNavFlight/inav #802: baro mode has gone???

jiekchenid, 2016-11-16:
baro mode 被 替代成了nave althold了嘛？？？现在定高模式需要GPS来执行了，BARO不在起作用了嘛。。。有没有纯靠 BARO起作用的模式

DzikuVx, 2016-11-16:
There is only ALTHOLD mode and this is per design.

closed 2016-11-16 by DzikuVx

oleost, 2016-11-16:
English?

And baro is replaced with althold mode, which also holds altitude but uses data from the inertial estimator instead of only relying on barometer. Its should work without GPS.

jiekchenid, 2016-11-16:
@oleost   but  the INAV 1.3    haven't    baro mode   only have nav althold   need gps

digitalentity, 2016-11-16:
@jiekchenid NAV ALTHOLD doesn't need GPS, only barometer sensor.

jiekchenid, 2016-11-16:
and  i can't  turn on  nav althold mode   i don't know why

oleost, 2016-11-16:
Probably because you have failed to calibrate the accelerometer. Please take you time and read trough the wiki. 

iNav is not cleanflight and there are some difference. You need to read the wiki.

jiekchenid, 2016-11-16:
@oleost   thanks  iread wiki now its working  but another problems  why  navposhold  mode   can not armed???????

为什么设置了nav poshold   电机就无法解锁了，，该模式下也无法激活？

juzhen698, 2016-11-29:
nav poshold 包括其他用到GPS的模式都要在能收到卫星，而且能定位后才能解锁。

liguo-git, 2017-02-20:
你的解锁了吗？我的也解不了锁，六面校准，油门行程最大最下也设置了也行。