iNavFlight/inav #812: Profile change bug

hali9, 2016-11-21:
How many Profile is in INAV. In cleanflight are 3 profile.
But when I change in inav-configurator to profile 3 then I get profile 1.
In CLI set profile to 2 then I get profile 0.

digitalentity, 2016-11-21:
Unable to reproduce on INAV 1.4-RC1

hali9, 2016-11-22:
On INAV 1.4-RC1 still there is a problem.
I use inav_1.4.0_CC3D.hex, maybe target is significant?

digitalentity, 2016-11-22:
This is a target-specific bug indeed.
We have a flash-size-specific `MAX_PROFILE_COUNT` definition:
```
#if FLASH_SIZE <= 128
#define MAX_PROFILE_COUNT 2
#else
#define MAX_PROFILE_COUNT 3
#endif
```

Not sure why we need this though.

digitalentity, 2016-11-22:
Looks like it has been recently introduced in https://github.com/iNavFlight/inav/commit/46fee424be7c79fff8427fdd1f42efae71f09624. @martinbudden do you remember why? This doesn't seem to affect binary size (at least not much).

PR #814 (2016-11-23): Change MAX_PROFILE_COUNT to 3 on all targets
Fixes #812
closed 2016-11-27 by digitalentity