iNavFlight/inav #813: baro problem in 1.4.0

mmormota, 2016-11-21:
# INAV/SPRACINGF3 1.4.0 Nov 18 2016 / 15:34:15 (f0de3a8)
board: FuryF3 with MS5611

The problem: the sensor chart shows something like the derivative of the altitude.
I mean if I move the quad 1m higher, there is an S like curve, first up then down wave, and the line returns to the zero value. If I move it back, again the S like curve but first down then up
Probably a bug in the low pass algo.

set baro_hardware = 3 doesn't help.

There are always 17 i2c errors at startup, no more errors after start.

# status
System Uptime: 77 seconds, Voltage: 118 * 0.1V (3S battery - OK), System load: 0.04
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000, BARO=MS5611
Cycle Time: 1005, I2C Errors: 17, config size: 2024

# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 501: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 501:  2          PWM_INIT_DONE 
 511:  3       EXTRA_BOOT_DELAY 
1209:  9         GYRO_DETECTION  (6, 0, 0, 0)
1511: 10          ACC_DETECTION  (7, 0, 0, 0)
1525: 11         BARO_DETECTION  (3, 0, 0, 0)
1696: 12          MAG_DETECTION  (1, 0, 0, 0)
1696: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
1696:  4       SENSOR_INIT_DONE 
2459:  8           SYSTEM_R

stronnag, 2016-11-21:
It would help if you 

* posted a link to your CLI dump; and 
* told us which build you're really using, because it's unlikely to be _my_ development build of Nov 18 2016 / 15:34:15 (f0de3a8). 

You probably mean `INAV/FURYF3_SPIFLASH 1.4.0 Nov 21 2016 / 00:15:30 (3275fc8)`

mmormota, 2016-11-21:
It's one iNav configurator offers as online download version.  

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/FURYF3_SPIFLASH 1.4.0 Nov 21 2016 / 00:15:30 (3275fc8)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
[... 386 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 0
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2016-11-22:
This is in fact intended behavior. Altitude is forced to be zero until the board is armed for the first time. This is intentional to ensure that home altitude is zero. Try re-testing after arming/disarming the board.

digitalentity, 2016-11-23:
Not a bug.

closed 2016-11-23 by digitalentity