iNavFlight/inav #830: Make initial phase of RTH configurable

digitalentity, 2016-11-28:
Currently RTH sequence is designed to go to (or avobe) desiged RTH altitude prior to turning home.
 
As suggested on RCG:
>  there's no reason why there could not be a setup option where the user chose the behaviour on FS in advance, eg turn immediately or climb to RTH altitude before turning.

Suggested option `nav_rth_initial_climb` with option to set `ON`/`OFF`. When set to `ON` - current behavior is executed, if set to `OFF` - UAV will turn and fly home instantly climbing/descending to target altitude on the go.

gbine, 2016-11-28:
What about integrating this with the variable 'nav_rth_alt_mode' ?  You could add  '_AFTER_TURN' to all the options except CURRENT to indicate desired behavior... 
EXTRA_AFTER_TURN, FIXED_AFTER_TURN, MAX_AFTER_TURN, AT_LEAST_AFTER_TURN indicating to turn first then climb.
For easy readability, the existing options could be changed by adding '_BEFORE_TURN'.
Taking it even further, if it makes sense and one coded it, there could be  '_DURING_TURN' to indicate a 'turn while climbing' action.

Regardless of how you do this, I  think this is a good idea.

digitalentity, 2016-12-06:
I believe it will be more confusing for end users. I'd rather keep one setting per behaviour modifier. Currently `nav_rth_alt_mode` defines the way RTH altitude is chosen. I don't want it to control something else. It will also make the code less readable.

closed 2016-12-12 by digitalentity