iNavFlight/inav #841: Failsafe recovery timeout

digitalentity, 2016-11-29:
Failsafe procedure is aborted as soon as RC link is restored. 
While is this good for common "LAND" failsafe, this is not good for RTH failsafe - might cause RTH to jump back and forth.

Suggesting to add a timeout for clearing the failsafe condition when RC link is restored.

digitalentity, 2016-11-29:
Currently we have a PERIOD_RXDATA_RECOVERY timer which is not enough. When failsafe procedure is RTH we probably want a few seconds of RTH executed correctly.

oleost, 2016-11-29:
Got mixed feelings here.

If it's only example 2s delay then sure maybe it does not hurt.

But I would not like to be locked out longer than necessary.

I actually experienced this one time, simple turning if the tx completely will save you.

digitalentity, 2016-11-29:
Two separate people approached me with this bug today. Maybe we can have a minimal 0.5s timeout by default with an option to increase up to several seconds (if anybody needs it).

oleost, 2016-11-29:
Sounds good.

But is it possible to increase PERIOD_RXDATA_RECOVERY  ?

Then people who know they setup they failsafe properly can increase this timer instead of adding a new 2. timer?

digitalentity, 2016-12-06:
PR in place, closing.

closed 2016-12-06 by digitalentity