iNavFlight/inav #844: Unintended In-Flight Disarm too easy with valid stick movements

dbechth, 2016-11-29:
I have been flying using ALTHOLD and POSHOLD for a couple months now and in that time I have now unintentionally disarmed my quad twice while trying to descend quickly and yaw left. Right, I get it, "don't do that", but the issue is that in both cases when flying I'm not thinking oooh I better stay out of that zone that would disarm the quad several hundred feet in the air.  Perhaps there is a feature already that would let you configure more complex stick movements to accomplish the arm and disarm function? If not I would like to propose a configuration option for this.

digitalentity, 2016-11-29:
Agreed. We probably should change stick combos for arming/disarming.

dbechth, 2016-11-29:
Additionally, I am fairly skilled in software development so if this is something of interest to others id be happy to help with this. I am currently working on setting up eclipse but haven't tried to build on my system yet.

digitalentity, 2016-11-29:
@dbechth welcome to the dev team then ;)

Unintended disarm is something that we must prevent, however we must do it carefully to avoid complications - i.e. using right stick will result in roll/pitch angle change.

Safest way is to arm/disarm by a switch though.

dbechth, 2016-11-30:
That's a good point, I didnt really think about the potential roll pitch change when using the right stick. Ill have to give this some more thought.

martinbudden, 2016-11-30:
> Safest way is to arm/disarm by a switch though.

Sort of true. It's possible to accidentally disarm by knocking the switch though, or moving the wrong switch by accident.

Part of the problem is transmitter design. I'd like to see a transmitter with "shoulder buttons" (like on toy quad transmitters). Then you could configure it so that you had to press both shoulder buttons simultaneously to arm/disarm. Or have it so that you had to double-click a shoulder button to arm/disarm.

oleost, 2016-11-30:
@Martinbudden You can almost do what with iNav already.

Either you need to use two channels for arming and then use the "and" function for modes. Or if you have an opentx transmitter you can always program it on the TX itself.

martinbudden, 2016-11-30:
You can do it with a transmitter with switches. But what I would actually like is "shoulder buttons" on the transmitter, like on toy quads, see picture.
[image]

digitalentity, 2016-11-30:
We need an INAV TX based on nRF24 code with telemetry and such
:laughing:

martinbudden, 2016-11-30:
That's a good idea!

I actually like the ergonomics of the toy transmitters. It's just a pity the pots are so bad.

dbechth, 2016-11-30:
How about the option for a two stage disarm?
Maybe something like:
Left Stick: Low Throttle & Yaw left. (This would put the flight controller into pre-disarm which would render the right stick unavailable for Roll and Pitch control)
Then 
Right Stick: Pitch Back & Roll Right

The advantage to this would be that this is a pretty uncommon stick combination and it would require that the stick combo happened in this specific order.

dbechth, 2016-11-30:
> Safest way is to arm/disarm by a switch though.

I've considered using that, however having only a 6Ch TX I'm already wishing I had more aux channels and would hate to dedicate 1 of my 2 just for arm and disarm.

stronnag, 2016-11-30:
As someone who flys iNav, Cf and βF, I appreciate having the same stick disarm sequence in all of them. Muscle memory works, especially for emergency disarm.   Whatever you do, please leave the possibility of remaining legacy compatible (even if it's a compile time option).

dbechth, 2016-11-30:
> Whatever you do, please leave the possibility of remaining legacy compatible (even if it's a compile time option).

Agreed. I'm not here to upset current users by removing functionality. I aim only to help prevent others from the disappointment I have experienced twice by accident. :)

oleost, 2016-11-30:
One possible solution is to allow the auto arm with no disarm on multirotors aswell.

This would of course require you to use motor stop feature. And step out of althold to actually stop motors.

digitalentity, 2016-12-01:
> I've considered using that, however having only a 6Ch TX I'm already wishing I had more aux channels and would hate to dedicate 1 of my 2 just for arm and disarm.

You don't have to dedicate a whole channel to arming - you can multiplex it with flight modes:
[image]

Along with this you should configure a mix on your TX to have Ch5 at min when 2-pos switch is off and when it's on - ch5 is set according to 3-pos flight mode switch to chose between Angle, Horizon and Acro.

dbechth, 2016-12-01:
@digitalentity 
In my case it isn't practical to use an arm switch or multiplex. I'm limited in what I can do because both CH5 and CH6 are only two position switches on my TX (on/off). If I set my modes this way I would only be able to chose a total of two flight modes one with CH5 and one with CH6. I'd love to just get a new radio, but that wont be in my hobby budget for awhile.

As always I really appreciate the help and suggestions. I am still pretty green in the multirotor hobby and have a lot to learn.

I have found the arming code and added my idea for the two stage disarm. If it works I'll move on to making it configurable and submit for review. I hope to test it out tonight!

too-many-projects, 2017-01-17:
How about having the standard disarm work at or near ground level. But when at altitude, require the disarm combo be performed twice within a short time period. This should stop accidental disarming and also be intuitive for people used to the normal disarm combo. If you forget you need to disarm twice it doesn't matter as the natural reaction would be to try the disarm again. I think it would be instinctive to try disarming a second time if it didn't work the first time.

dbechth, 2017-01-17:
@too-many-projects Interesting idea. I like the intuitive nature of that solution. Any thoughts on what happens with no baro? Would the feature just be unavailable?

too-many-projects, 2017-01-17:
Without a baro maybe fall back to using an average throttle value to decide on disarm requirement. A low average throttle requiring a single disarm. Whereas a high average throttle requires a double disarm. High average throttle is likely normal flight, low average throttle is likely landed or at rest state.

DzikuVx, 2020-04-14:
stick armimg was removed, not relevant

closed 2020-04-14 by DzikuVx