iNavFlight/inav #847: Airplane launch assistant improvements

digitalentity, 2016-11-30:
Fix non-detected launches when throwing a flying wing by the wingtip.

digitalentity, 2016-12-05:
@oleost this will probably have to wait for after the New Year. Too much testing is required.

oleost, 2016-12-05:
No problem. Just postpone to 1.6
Remember you can always ask me to do some test.

But of course, now its christmas / new year and the days are short so its limited whats time left to do some testing.

digitalentity, 2016-12-05:
I'd rather keep it in 1.5 for now - maybe I'll have the time to implement new logic

digitalentity, 2016-12-07:
Yee-ha! Experimental #878 done!

flounderscore, 2016-12-07:
How are you supposed to trigger the launch?

digitalentity, 2016-12-07:
It's not for triggering the launch, it's detecting the launch. You just throw the airplane and INAV takes over helping you to reach stable climb.

oleost, 2016-12-07:
@flounderscore  https://github.com/iNavFlight/inav/wiki/8.-Navigation-modes#launch---airplane-launch-assistant

closed 2016-12-12 by digitalentity