iNavFlight/inav #848: Gas engines support

DzikuVx, 2016-11-30:
To satisfy gas engine fixed wings we would need some changes in terms of output channels:

1. Throttle channel with PMW servo signal
1. Killswitch channel. 1000us means engine is disabled, 2000us engine is enabled. Probably it makes sense to pass ARM state here
1. heater channel. 1000us - no heating, 1500 - heating, 2000 - full heat for engine start. It also would be nice to enable heating when throttle drops below threshold

killswitch and heater ranges should be configurable.

There is also a matter how to arrange those additional channels for gas engines. The way I see it, M1 and M2 and not used here. Maybe the can be reused together with with Servo 5 (outout 7) that currently passes throttle using "servo PWM" frequency.

Thought?

driver33, 2016-11-30:
Description sounds ok, IMHO in such configuration its wise to remove all "Motors" configuration to get rid off occupying  pwm outputs (0,1) as its now set 2 motors on all configurations.

MrScott66, 2018-12-01:
What is the latest with gas planes and iNav?   I want to power a servo off channel one to run a glow engine.  A channel for a kill switch and another for a glow driver can be handled other ways.  Is it or could it be possible to run a servo off channel 1?  Thanks

digitalentity, 2018-12-01:
As long as you set motor protocol to PWM and motor rate to 50Hz - you can use a servo on motor outouts.

MrScott66, 2018-12-08:
Ive got that part working and I can control the servo on the motors tab with the slider, but I can't get any movement on the servo from the radio.  The receiver tab shows that the receiver is receiving the signal just like the other channels.  Any ideas?  Thanks!

shellixyz, 2018-12-08:
@MrScott66 The motor channel is disabled (set to min_command) when the FC is disarmed

MrScott66, 2019-01-21:
Ok, I get to start over.  I had everything working but my board let out the magic smoke and I updated Inav.  Now there is no PWM option under the ESC protocol.  There is just ESC protocols.  Using bushed or standard makes my little servo want to go full deflection and self destruct.   Also, I'm getting no servo movement for servo outputs 3 thru 6, even if it is armed and the receiver tab shows the signals are being received.  

Thanks for the help

MrScott66, 2019-01-21:
I figured out the servos other than the throttle channel.  There is no 5V going to the output.  so my only issue is getting the throttle output servo, not ESC, compatible.

Thanks again

DzikuVx, 2020-12-11:
All of thet can be implemented with INAv mixer, closing

closed 2020-12-11 by DzikuVx

puertobri, 2025-11-06:
Hello, I'm looking for a way to connect a servo to channel one or two of my flight controller to power a combustion engine. I've been reading but haven't found how to do it.