iNavFlight/inav #851: Flip32 F4 Tricopter configuration doesn't work

digitalentity, 2016-12-01:
Reported by a user who replaced Flip32 with Flip32-F4 on his Tri.

oleost, 2016-12-01:
Where are servo output on Flip32 F4?

If it the same as F3 it's on pin 7 and that's different from the pin 1 he is used from on naze32

DzikuVx, 2016-12-01:
Ha, this is a very good question, since I have no idea :(
Let's find it, but probably we will have to move it from where it is right now

DzikuVx, 2016-12-01:
Yes, we have a bug, servo outouts in non-airplane configurations are not initialized at all

digitalentity, 2016-12-01:
Let's put this on hold for now until @martinbudden does his big thing with config access macros.

martinbudden, 2016-12-01:
Hope to get the config access macro change in next week.

digitalentity, 2016-12-12:
Fxied by https://github.com/iNavFlight/inav/pull/872

closed 2016-12-12 by digitalentity