iNavFlight/inav #854: Altitude control for RTL on an airplane

digitalentity, 2016-12-01:
From @marbalon
> I have a question for RTL for fixedwings... I know automatic modes control altitude and fallow nav_rth_alt_mode settings, but is there any option co control alt hold manually ?
> sometimes I fly away and when I return I switch RTL, but want to climb up/or decrease altitude during fly
> is there any option to dissable alt hold for RTL ? I want to manage altitude manually or just use alt hold switch to hold it...

marbalon, 2016-12-01:
Maybe the easiest way is add flag rtl_control_altitude =ON/OFF... easy to understand for users.

sppnk, 2016-12-01:
IMHO,  RTH should be a easy-clean-clear mode, a bomb proof thing. I guess it shouldnt be too complex.

closed 2016-12-26 by digitalentity