iNavFlight/inav #860: inav 1.4  telemetry  BUG _help me

wolflonezx11, 2016-12-02:
hi 

inav v1.4
naze32 ver6
frsky d8r 

Inav 1.3 works normally.
Ver 1.4 up the transmitter contents
1. SWR
2. RSSI
3. A1
4. A2
The following contents are not displayed.

# dump

# version
# INAV/NAZE 1.4.0 Nov 28 2016 / 10:55:47 (6bf7a91)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature SOFTSERIAL
feature FAILSAFE
feature TELEMETRY
feature RSSI_ADC
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
beeper LAUNCH_MODE

# map
map AETR1234

# serial
serial 0 1 115200 38400 0 115200
serial 1 0 115200 38400 0 115200
serial 30 0 115200 38400 0 115200
serial 31 4 115200 38400 9600 115200

set looptime = 2000
set i2c_overclock = OFF
set gyro_sync = OFF
set gyro_sync_denom = 2
set acc_task_frequency = 500
set attitude_task_frequency = 250
set async_mode = NONE
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set rc_smoothing = ON
set input_filtering_mode = OFF
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 1000
set motor_pwm_rate = 400
set motor_pwm_protocol = STANDARD
set fixed_wing_auto_arm = OFF
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set small_angle = 25
set reboot_character = 82
set gps_provider = UBLOX
set gps_sbas_mode = NONE
set gps_dyn_model = AIR_1G
set gps_auto_config = ON
set gps_auto_baud = ON
set inav_accz_unarmedcal = ON
set inav_use_gps_velned = ON
set inav_gps_delay = 200
set inav_gps_min_sats = 6
set inav_w_z_baro_p =  0.350
set inav_w_z_gps_p =  0.200
set inav_w_z_gps_v =  0.500
set inav_w_xy_gps_p =  1.000
set inav_w_xy_gps_v =  2.000
set inav_w_z_res_v =  0.500
set inav_w_xy_res_v =  0.500
set inav_w_acc_bias =  0.010
set inav_max_eph_epv =  1000.000
set inav_baro_epv =  100.000
set nav_disarm_on_landing = OFF
set nav_use_midthr_for_althold = OFF
set nav_extra_arming_safety = ON
set nav_user_control_mode = ATTI
set nav_position_timeout = 5
set nav_wp_radius = 100
set nav_max_speed = 300
set nav_max_climb_rate = 500
set nav_manual_speed = 500
set nav_manual_climb_rate = 200
set nav_landing_speed = 200
set nav_land_slowdown_minalt = 500
set nav_land_slowdown_maxalt = 2000
set nav_emerg_landing_speed = 500
set nav_min_rth_distance = 500
set nav_rth_tail_first = OFF
set nav_rth_alt_mode = AT_LEAST
set nav_rth_altitude = 1000
set nav_mc_bank_angle = 30
set nav_mc_hover_thr = 1500
set nav_mc_auto_disarm_delay = 2000
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_bank_angle = 20
set nav_fw_climb_angle = 20
set nav_fw_dive_angle = 15
set nav_fw_pitch2thr = 10
set nav_fw_roll2pitch = 75
set nav_fw_loiter_radius = 5000
set nav_fw_launch_accel = 1863
set nav_fw_launch_detect_time = 40
set nav_fw_launch_thr = 1700
set nav_fw_launch_motor_delay = 500
set naw_fw_launch_timeout = 5000
set naw_fw_launch_climb_angle = 10
set serialrx_provider = SPEK1024
set spektrum_sat_bind = 0
set telemetry_switch = OFF
set telemetry_inversion = ON
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set gyro_lpf = 42HZ
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 50
set imu_dcm_kp_mag = 10000
set imu_dcm_ki_mag = 0
set pos_hold_deadband = 20
set alt_hold_deadband = 50
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set servo_center_pulse = 1500
set servo_pwm_rate = 50
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = RTH
set rx_min_usec = 885
set rx_max_usec = 2115
set acc_hardware = 0
set baro_use_median_filter = ON
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SERIAL
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0
set acczero_x = 0
set acczero_y = 0
set acczero_z = 0
set accgain_x = 4096
set accgain_y = 4096
set accgain_z = 4096

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# dump profile

# profile
profile 0

# aux
aux 0 0 0 900 900
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

set nav_alt_p = 50
set nav_alt_i = 0
set nav_alt_d = 0
set nav_vel_p = 100
set nav_vel_i = 50
set nav_vel_d = 10
set nav_pos_p = 65
set nav_pos_i = 120
set nav_pos_d = 10
set nav_posr_p = 180
set nav_posr_i = 15
set nav_posr_d = 100
set nav_navr_p = 10
set nav_navr_i = 5
set nav_navr_d = 8
set deadband = 5
set yaw_deadband = 5
set throttle_tilt_comp_str = 0
set flaperon_throw_offset = 250
set flaperon_throw_inverted = OFF
set gimbal_mode = NORMAL
set fw_iterm_throw_limit = 165
set mode_range_logic_operator = OR
set default_rate_profile = 0
set mag_declination = 0
set mag_hold_rate_limit = 90
set p_pitch = 40
set i_pitch = 30
set d_pitch = 23
set p_roll = 40
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_level = 20
set i_level = 15
set d_level = 75
set max_angle_inclination_rll = 300
set max_angle_inclination_pit = 300
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

oleost, 2016-12-02:
Looks like e3355d591982c8105198a25a9efe9d27844f2a1f removed Frsky telemetry from F1 targets aswell.

Was this intentionall? If so we need to update Wiki and hardware map.

digitalentity, 2016-12-02:
Hmm... True. I'll check if it was intentional.

martinbudden, 2016-12-02:
It was taken out to save ROM space. I've just tried putting it back in and NAZE ROM overflows by 816 bytes. I don't think there is an easy way of saving that much ROM to allow it to be put back in, so we need to update the wiki.

oleost, 2016-12-02:
Updated Wiki, Hardware map and 1.4 Release notes.

@wolflonezx11 Seems like we had to remove Frsky due to size issues on iNav 1.4

Your solution now would try and compile your own version with FrSky telemetry instead of LTM telemetry.

martinbudden, 2016-12-02:
@oleost , thanks for doing the documentation updates.

stronnag, 2016-12-02:
Or just compile with gcc 6.2 :), ltm & frsky:
````
arm-none-eabi-size ./obj/main/inav_NAZE.elf
   text	   data	    bss	    dec	    hex	filename
 127492	   1384	  17520	 146396	  23bdc	./obj/main/inav_NAZE.elf
````

wolflonezx11, 2016-12-03:
Thank you
Check the firmware and insert it
I'll leave a message after.

DzikuVx, 2016-12-15:
@wolflonezx11 so does it works now? Can this ticket be closed

wolflonezx11, 2016-12-16:
Resolved.
Thank you.

closed 2016-12-16 by DzikuVx