iNavFlight/inav #861: Heading is drifting to set position (F3 EVO with MAG)

Ekci, 2016-12-04:
After first failed GPS POS HOLD test I tried my F3 EVO board on the desk and realized that the heading is drifting to some set position all the time, degree per second or so.

Problem existed even when powering from USB (no noise from BEC or ESC). Raw data in sensor tab looks OK, no drift there. Any idea what's wrong?

Dump:
`# dump

# version
# INAV/SPRACINGF3EVO 1.4.0 Nov 28 2016 / 11:00:40 (6bf7a91)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature RX_PPM
feature MOTOR_STOP
feature GPS
feature FAILSAFE
feature RSSI_ADC
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
[... 387 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 55
set yaw_rate = 25
set tpa_rate = 20
set tpa_breakpoint = 1650

# `

stronnag, 2016-12-04:
You mag calibration values are bizarre. I don't use the on board (as I value my machines), but my on board mag appears to calibrate reasonably sanely:
````
set magzero_x = -100
set magzero_y = 464
set magzero_z = -231
````
Compare to your dump:
````
set magzero_x = -625
set magzero_y = -1
set magzero_z = -8777
````
To me it looks as if your calibration is bad, or the hardware is faulty.

Ekci, 2016-12-04:
Yeah, that's the result of one of many fast calibration. 
I did proper one:
> set magzero_x = 287
> set magzero_y = 407
> set magzero_z = -1089

closed 2016-12-14 by digitalentity

martindb, 2017-04-15:
Ekci how do you solve your problem? I have a sp racing f3 clone (banggood), and magnetometer is not holding the heading.... just drifting all the time. Tested with different versions and also cleanflight, same behavior. No magnetic fields near the board.
https://youtu.be/Fy_t_NbFGRY

Ekci, 2017-04-29:
Sorry for this late answer. Electromagnetic interference from power distribution board was my main problem, nothing to do with software. I solved id by placing grounded copper plane between both PCBs.