iNavFlight/inav #866: Bug:if UART2 was opened then the motor can't move.

Linjieqiang, 2016-12-06:
Recently,I use STM32F1 chip to make my mini brushed copter.But I found if I opened the GPS with UART2,the motor M2 & M4 can not move.I try to use other timer but it didn't work.So I think maybe some bugs in the timer and uart2.

[image]

digitalentity, 2016-12-06:
@Linjieqiang what target?

Linjieqiang, 2016-12-06:
My new define target.
[image]
[image]
[image]

digitalentity, 2016-12-06:
F1 or F3?

Linjieqiang, 2016-12-06:
F1.But I think CC3D & NAZE have the same problem.

digitalentity, 2016-12-06:
CC3D and NAZE doesn't have this problem

Linjieqiang, 2016-12-06:
But what's wrong with my target?I haven't found the problem.

digitalentity, 2016-12-06:
Ok, this is not affecting other targets by coincidence.

NAZE target have PWM3 and PWM4 channels dedicated for PWM receiver, and those are shared between UART2 and PWM inputs. I'll fix the underlying issue. Stay tuned.

Linjieqiang, 2016-12-06:
But what should I do to let my new mini copter fly?I am a junior programmer.

digitalentity, 2016-12-06:
@Linjieqiang have a look at 34ae508b1cc720f38164c38d2818bbfbfde8c057

Linjieqiang, 2016-12-06:
OK.Let me have a check.

Linjieqiang, 2016-12-06:
I just had a try and it worked!Amazing!Thanks @digitalentity

closed 2016-12-06 by Linjieqiang

digitalentity, 2016-12-06:
@Linjieqiang no problem. It's a long lurking issue. We have only a few F1 targets, so we never noticed before.

Linjieqiang, 2016-12-06:
OK.I closed the issue.