iNavFlight/inav #874: Problem with Naze32 rev 6 - external mag not detected in inav 1.4

fightersky, 2016-12-07:
Hi 
I'm trying to setup external mag in inav, but after  I  set align_mag = CW270FLIP in CLI, magnetometer is greyed out and not detected from then on.  I also have 10 I2C errors. I've installed external mag in cleanflight and it works perfectly no errors or detection issues. 
This is my first quad build and I'am no expert.
I also couldn't  calibrate accelerometer using 6 point method, had 4096 error in CLI.
 

Anyone else having similar issues?

Naze 32 rev 6 barometer / no internal mag- purchased from hobbyking

cheers

MarcoPTG, 2016-12-07:
Working great with ublox m6 neo ext mag. For me no issue. Re check the wiring.

digitalentity, 2016-12-07:
Please provide output of `dump` and `bootlog` commands.

fightersky, 2016-12-07:
plug wrong wire in now naze32 wont boot up, now looking at youtube to unbrick

fightersky, 2016-12-07:
how do we output dump and bootlog commands?

fightersky, 2016-12-07:
[image]
[image]
[image]
[image]

the magnetometer is now recognised, god only knows what's going on, but it's  not working very well even after calibration. Moves in the correct direction, but if I quickly move in opposite direction its very slow and gives false readings. I have  given mag at least 10cm distance from noise. I'am currently using single external board mag which is CW0. I previously stated CW270FLIP but this was wrong that was my other external mag with gps,  board orientation using HMC5883L is CW0.  

cheers

fightersky, 2016-12-07:
no answer the question??

fightersky, 2016-12-07:
that's because the naze 32 rev 6 with external mag doesn't work correctly. Other I see have raised this issue before. Should warn others so they don't spend 3 hrs stuffing around with it

fightersky, 2016-12-07:
going to cleanflight

fightersky, 2016-12-07:
great help

digitalentity, 2016-12-07:
This is not a paid support service, you shouldn't expect instant responses and instant problem solving. 

The problem with Naze32 Rev6 is known and related to hardware. Even with other firmwares your magnetometer will eventually fail on you but that will more likely happen in the air.

closed 2016-12-07 by digitalentity