iNavFlight/inav #880: PWM RC not working on CC3D

TradeCrafter, 2016-12-08:
https://www.youtube.com/watch?v=zT57Z4heyGw

you can see the oscillating issue on the video, the CC3D flashed with CleanFlight works fine, same with LibrePilot and OpenPilot...

so the only thing i can think of is a bug...

RADIO: FlySky FS-i6
RECEIVER: FS-iA6  6ch

digitalentity, 2016-12-08:
Dump? Bootlog?

oleost, 2016-12-08:
@TradeCrafter Just want to confirm with you that issues is confirmed and they are looking into it.

digitalentity, 2016-12-08:
@TradeCrafter can't reproduce the issue on my FrSky system using INAV-1.4 and all 6 PWM inputs. Please provide a config dump.

TradeCrafter, 2016-12-08:
it was not an hookup on first test, i mounted the F-450, flew and crashed, mounted again... the same issue, reviewed all the setup, wirings, solderings, i've also looked for possible interference, separate wires from power sources, magnetic fields, etc...

i don´t have any logs, right now i got this CC3D flashed with cleanflight, i needed to fly safe, hope to get back on iNAV when arrives my new i-BUS receiver.

digitalentity, 2016-12-08:
@TradeCrafter FYI, I have a FlySky system in transit. Will re-test and try to confirm the bug.

Also can you answer a few questions?

1. What version of INAV were you using?
1. What variant of CC3D firmware were you using (currently there are CC3D, CC3D_PPM1)
1. Did you configure OneShot, softserial or any other features?
1. If you have a CLI `dump` output - it would also be quite good to analyse.

TradeCrafter, 2016-12-08:
1,2= tryied: "inav_1.2.0_CC3D_OPBL.bin" and "inav_1.2.0_CC3D.hex"
3- no softserial, using telemetry on uart1, and GPS ublox on uart3, but i also set without this itens, only FC and r/c, still...
4- i dont have a CLI dump...  sorry

digitalentity, 2016-12-08:
Ok, I didn't try to test version **that** old (1.2 was relesed over 3 months ago and a lot of bugs was fixed since than). If  you have the time, can you try 1.4 and see if the bug persists?

TradeCrafter, 2016-12-08:
sorry my mistake, i updated my last comment, i try the last one for sure, this is the first one i tried...

TradeCrafter, 2016-12-08:
i think its not a matter of version... happens on both versions i tried.

digitalentity, 2016-12-08:
Ok, FlySky-specific then. This will have to wait until I get the hardware to test.

digitalentity, 2016-12-21:
Issue confirmed. @TradeCrafter will you be able to test the possible fix?

digitalentity, 2016-12-21:
See https://github.com/iNavFlight/inav/pull/956 for a fix

TradeCrafter, 2016-12-21:
i am using openpilot on the FC, but i can re-flash with inav again to try !!
just point me in the right directions ?
is the fix on a new FW release ?
i read "See #956 for a fix"  but just don't understand what to do, seems like any change needs to be done on some C file for handle with pwm capture...

i am happy to help man !

digitalentity, 2016-12-22:
@TradeCrafter I made you a hex (OPBL version doesn't fit into memory now). Please try this hex on your setup.

[inav_1.5.0_CC3D.hex.gz](https://github.com/iNavFlight/inav/files/668285/inav_1.5.0_CC3D.hex.gz)

TradeCrafter, 2016-12-22:
it worked like a charm, @digitalentity !! bug fixed!!  glad to help !!!

digitalentity, 2016-12-22:
@TradeCrafter thanks for testing!

TradeCrafter, 2016-12-22:
i'm having some trouble with my 915mhz 3dr telemetry, with older version of iNAV it worked... there's any change needs to be done in order to get it working ?

digitalentity, 2016-12-22:
What telemetry are you using? CC3D only supports LTM due to memory limitations

TradeCrafter, 2016-12-22:
china cheap ones, running SiK firmware, wich has on config utility the MAVLink or RAW DATA protocols, i read on iNAV wiki, about this forked firmware: https://github.com/stronnag/SiK-MSP

do you know by any means if theres is a chance to use it on my 3dr ?

stronnag, 2016-12-22:
You can run any protocol over it (MSP, LTM, Mavlink). The supplied firmware is optimised for mavlink framing, this does preclude using MSP/LTM. Lots more detail in the wiki https://github.com/iNavFlight/inav/wiki/iNavFlight-Missions

TradeCrafter, 2016-12-22:
ok my radios are exactly these: [https://pt.aliexpress.com/item/3DRobotics-400mW-Telemetry-Kit-915Mhz-915-Module-For-APM-APM2-5-2-6-Pixhawk-For-Quad/32251253239.html?spm=2114.13010608.0.0.KtgjVq&detailNewVersion=&categoryId=200001407](url)

i dont have the silabs debug adapter, i got an USB/TTL adapter, can i use it instead ?
this firmware of yours will work on my hardware ?
it needs to get bootloader updated also ?

stronnag, 2016-12-22:
The modified firmware will work. I have the 433Mhz version. You can use an USB/TTL to update the firmware. You do not need to update the bootloader.

TradeCrafter, 2016-12-22:
last question: can i use the config utility from 3D Robotics, for windows, to perform the flash ?

stronnag, 2016-12-22:
Sorry, no experience of Windows. The GUI from http://vps.oborne.me/3drradioconfig.zip should work on Windows and Linux, for flashing and configuration.

TradeCrafter, 2016-12-22:
this is the GUI for windows i use, just cant get to compile the firmware for the radios, using VM on Mint, the ec2tools just wont install... 
it will be the same bin/hex file for the 2 radios ? air and ground ?

TradeCrafter, 2016-12-22:
can i use the: radio~hm_trp.ihx  ?

stronnag, 2016-12-22:
Same firmware for both (it is entirely symmetrical)
http://www.seyrsnys.myzen.co.uk/3dr/radio~hm_trp.ihx or the file from github.

TradeCrafter, 2016-12-22:
managed to upload the firmware, and get both radios sync, green lights remains still and get response on local and remote radios, on the FC, i've put LTM on "ports" tab, UART1, marked "telemetry" on "configuration" tab, got red blinks, but no conection with inav trough 3dr... i using 56.700 baudrate on my radios

stronnag, 2016-12-22:
So can you post the iNav serial setup (serial in the CLI), and the 3dr setup?

TradeCrafter, 2016-12-22:
looks like a joke... i got all 4 ESCs beeping th first 3 beeps, but the longer beep it only came from 3 ESCs, one of them dont beep the second beep, it's okay before flahsing...  i need to go out now, later i will look furhter on telemetry... i just worried about this ESC... (simonk 30A, china red ones) 
:(

TradeCrafter, 2016-12-22:
MAAAAAAN !!!!! you made goo all across town here in Rio de Janeiro to buy another ESC, and when i get it into just doesn't work like the other one... so i flashed CopterControl firmware again and thats it !! ALL 4 ESCs and MOTORS ready to fly again... there´s something on the HEX you sent me...
and i got an different brand ESC for replace now.. i can't believe bro... the issue was on the firmware...

digitalentity, 2016-12-22:
Didn't you forget to enable ESC outputs in Configurator?

[image]

digitalentity, 2016-12-22:
PWM input is now fully functional, #956 is merged, closing this issue.

closed 2016-12-22 by digitalentity

TradeCrafter, 2016-12-23:
without enabling esc output, no one motors spin... enabling it works only 3 motors...

digitalentity, 2016-12-23:
This is strange. From my bench-tests all motor outputs work fine on CC3D+PWM RX, I'll need to test this more, thanks.