iNavFlight/inav #916: REVO: I2C error count is skyrocketing if BARO is enabled

digitalentity, 2016-12-13:


night-ghost, 2016-12-13:
mentioned earlier software I2C driver [published](https://github.com/night-ghost/ardupilot/blob/master/libraries/AP_HAL_REVOMINI/i2c_soft.cpp). Works fine even with damned MS5611 baro :)

digitalentity, 2016-12-13:
Tracked down to ebe5668073eedc0d1282323f7c1331596800e905. Looks like timeout of 40960 dummy cycles is longer than new timeout of 1ms.

digitalentity, 2016-12-13:
I2C error count seems to be growing fast only when the board is connected to USB.

digitalentity, 2016-12-13:
If I change timeout from using micros() to use a simple counter issue is resolved:
```
    timeUs_t fakeClock = 0;
    do {
        //const timeUs_t currentTimeUs = micros();
        i2cStateMachine(&busState[device], fakeClock++);
    } while (busState[device].state != I2C_STATE_STOPPED);
```

I wonder if some ISR is taking too long to execute. My first suspect is USB VCP code.

digitalentity, 2016-12-13:
I'll make a hacky fix using cycle counter, but we'll have to resolve the underlying issue some day.

digitalentity, 2016-12-13:
Ok, tried two things:

1. Keeping current `micros()`-driven timeout and increasing I2C_TIMEOUT up to 1 second - didn't help.
1. Switching to a cycle-count-style timeout https://github.com/iNavFlight/inav/pull/922/commits/cdb40a633a215f0416fc2c579f0cd9c20a484320 - fixes the issue.

I'm stuggling to understand why increasing the timeout doesn't help. @martinbudden, @ledvinap do you have any idea why this might be happening?

night-ghost, 2016-12-13:
Just try a 3rd way - not use micros() but compare time in ticks with SysTick value, It's enough to not too much delay. Not forget to multiply time by usTicks

digitalentity, 2016-12-13:
@night-ghost do you think that `micros()` might return incorrect values?

night-ghost, 2016-12-13:
yes this was in my case

ledvinap, 2016-12-13:
@digitalentity : That's strange, I would expect `micros()` to work now ... Cay you try with lover optimization - `-O2`/disable `-flto`?

night-ghost, 2016-12-13:
or try this version
```
uint32_t systick_micros(void)
{
    uint32_t fms, lms;
    uint32_t cycle_cnt;
    uint32_t res;
    do {
        // make sure systick_uptime() return the same value before and after
        // getting the systick count
        fms = systick_uptime();
        cycle_cnt = systick_get_count();
        lms = systick_uptime();
    } while (lms != fms);

#define US_PER_MS               1000
    /* SYSTICK_RELOAD_VAL is 1 less than the number of cycles it
       actually takes to complete a SysTick reload */
    res = (fms * US_PER_MS) +
        (SYSTICK_RELOAD_VAL + 1 - cycle_cnt) / CYCLES_PER_MICROSECOND;

    return res;
#undef US_PER_MS
}
```

ledvinap, 2016-12-13:
@night-ghost : `micros()` shall use this logic when not in interrupt context ...

night-ghost, 2016-12-13:
micros() in INAV
```
uint32_t micros(void)
{
    register uint32_t ms, cycle_cnt;
    do {
        ms = sysTickUptime;
        cycle_cnt = SysTick->VAL;
        /*
         * If the SysTick timer expired during the previous instruction, we need to give it a little time for that
         * interrupt to be delivered before we can recheck sysTickUptime:
         */
        asm volatile("\tnop\n");
    } while (ms != sysTickUptime);
    return (ms * 1000) + (usTicks * 1000 - cycle_cnt) / usTicks;
}

```
difference is small but it is

digitalentity, 2016-12-13:
There's definitely something wrong with `micros()` implementation. Using DWT (and multiplying the delay by CPU MHz) solved the problem. A hacky but working solution.

```
    CoreDebug->DEMCR |= CoreDebug_DEMCR_TRCENA_Msk;
    DWT->CTRL |= DWT_CTRL_CYCCNTENA_Msk;

    do {
        timeUs_t currentTimeUs = DWT->CYCCNT;
        i2cStateMachine(&busState[device], currentTimeUs);
    } while (busState[device].state != I2C_STATE_STOPPED);
```

night-ghost, 2016-12-13:
You can not trust anyone, even the system libraries. Miсros() in that form is copied from the Arduino, which has a fundamentally different interrupt handling

ledvinap, 2016-12-13:
@night-ghost: this version of micros shall work: https://github.com/iNavFlight/inav/blob/master/src/main/drivers/system.c#L106-L126

The one you posted won't work in interrupt context (probably both of them)

digitalentity, 2016-12-13:
Interesting thing is that timeout event in I2C state machine is triggered precisely after I2C_DELAY elapsed according to `micros()`. If `micros()` would jump, I would be seing larger delays, but I'm not.

night-ghost, 2016-12-13:
disabling interrupts before accessing multi-word data is much more reliable :)

digitalentity, 2016-12-13:
@ledvinap `-O2` or disabling `-lto` breaks I2C - baro is no longer detected.

digitalentity, 2016-12-13:
More tests:
Same code (with `micros()`).
I2C_TIMEOUT = 1000 (1ms) - BARO detected, I2C error rising fast
I2C_TIMEOUT = 10000 (10ms) - BARO not detected at all

digitalentity, 2016-12-13:
Changed `micros()` to `millis()` and adjusted the delay accordingly - works perfectly. Looks like `micros()` is seriously broken :scream:

night-ghost, 2016-12-13:
just add 2 helper functions - uint32_t ticks() which return DWT, and uint32_t time_since(uint32_t tick) which return by DWT time in microseconds from given tick - and forget all micros() and millis() for waiting needs

digitalentity, 2016-12-13:
That's probably the best thing to do at the moment, but that's hiding the actual problem, not solving it. The problem is that `micros()` returns incorrect values for some reason.

night-ghost, 2016-12-13:
If your code not uses broken micros() at all - this is not a problem :). That function itself is too heavy for just waiting needs

PS. Try to remove in micros() all about prio levels and simply disable interrupts before access to timer's variables

ledvinap, 2016-12-13:
@night-ghost : As @digitalentity said - it's not a good idea to mask actual problem ...

There is some errata for F1 processors, I2C must be handled immediately in some context. Using top priority group is one possibility. I implemented workaround in my local branch (higher priority for timer did break I2C), here is the commit: https://github.com/ledvinap/cleanflight/commit/5f9354fbefd199a3ef757f10a820c1109a10bb7c. Maybe it's related ...

ledvinap, 2016-12-13:
BTW: The idea of short-timespan timer function for delays good one.
One possibility is to let systick run to maximum period (24bit of ticks, ~0.2s) and adapt `millis()` use systick value in similar way as `micros()`.
I did it this way: https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/timer.c#L809, it assumes millis is called at least once per `micros()` overflow (my code uses general timer for micros, but systick should do)

night-ghost, 2016-12-13:
> As @digitalentity said - it's not a good idea to mask actual problem ...

Remove affected by cancer organs is not masking - it is the treatment.

night-ghost, 2016-12-13:
>adapt millis() use systick value in similar way as micros().

the worst idea I've ever heard. If we have not working micros() then let's break millis() too, sure?

ledvinap, 2016-12-13:
@night-ghost : I'm quite confident that `micros()` does work ... at least the one in master. And if it doesn't, it can be fixed
The idea was to get very fast (single-read) 24bit tick counter for operations where 200ms is enough.

digitalentity, 2016-12-14:
Whoohoo! Fixed.
Looks like `micros()` wasn't the root cause. The root cause was not waiting for `STOP` condition to clear which caused the next transaction on the bus to start too early. Anyway, I'll be using new `ticks()` function (based on DWT) to check for timeout - it has much smaller overhead than `micros()` and all our CPUs have DWT capability.

digitalentity, 2016-12-14:
As a side-effect I've done an upgrade to CMSIS v5.0 - works fine.

digitalentity, 2016-12-14:
Fix merged. Closing.

closed 2016-12-14 by digitalentity