iNavFlight/inav #921: Question: why delay() is ONLY delay?

night-ghost, 2016-12-13:
and not task switch to scheduler? I don't know how it here but profiling of Arducopter shows 67% of CPU time just in delay()

digitalentity, 2016-12-13:
Because scheduler is cooperative, not preemptive. There is no such thing as task switch.

night-ghost, 2016-12-13:
What a speed of closing! :) JFYI, even cooperative scheduler CAN switch tasks  -  https://github.com/night-ghost/ardupilot/blob/master/libraries/AP_HAL_REVOMINI/Scheduler.cpp#L562

digitalentity, 2016-12-13:
This is not an APM, our scheduler assumes task is never interrupted. 
However having a `yield()` for long running tasks might be a good thing. Re-opening :)

night-ghost, 2016-12-13:
that code ALSO is not a APM because it is just mine hack, APM has the same problem. See: task calls delay() so scheduler knows than this time is free for another tasks. All that it need is a maximal time in the task before next yield() in tasks to not miss time

night-ghost, 2016-12-13:
so all low-priority tasks such as telemetry, LEDS and so on can be scheduled in free time

digitalentity, 2016-12-13:
That's already happening with current code - low priority tasks are scheduled only if enough free time is available.

night-ghost, 2016-12-13:
how much % of time spent in delay() ?

night-ghost, 2016-12-13:
Well then here is much better than in the APM

digitalentity, 2016-12-13:
Usually tasks here are designed in such a way that they don't have to wait or keep their own time-driven scheduling for slow events.

closed 2023-10-27 by DzikuVx