iNavFlight/inav #926: OSD - navigation-specific indications

digitalentity, 2016-12-14:
For best user experience we need to add a few more things to OSD:

- UAV heading indicator
- GPS coordinates
- HOME distance
- HOME direction
- Next waypoint direction (for WP mode)
- Total trip distance

Schalonsus, 2017-01-31:
- Efficiency Display (Wh/km)

Maybe there could occur some errors:
- High altitude support (>9999m)
- Long distance support (>99999km)
- Negative altitude support

Dont know if it is implemented yet:
- Summary screen (Flight Time; Total Distance; Max Distance; Max Altitude; Max Speed; mAh used; Max Amps

marbalon, 2017-02-01:
Most of this is already done, will commit after this weekend.

Schalonsus, 2017-02-07:
Is there an inbuilt OSD menu for adjusting PIDs and other parameters like with the MWOSD firmware for the MinimOSD?
This makes tuning in the field alot easier. Not every aircraft has a bluetooth module or access to the usb port and the right device in the field.
https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/User_Guide.md#mw-osd-quick-reference-guide---mwosd-r16
https://youtu.be/pd_PXnG_PkI

martinbudden, 2017-02-10:
@digitalentity , @marbalon , what's the status of this issue?

digitalentity, 2017-02-10:
@martinbudden no idea, but this is the last PR that's holding RC1 release.

marbalon, 2017-02-10:
I have most of this codded:

UAV heading indicator - done
GPS coordinates - done
HOME distance - done
HOME direction - done
Next waypoint direction (for WP mode) - missing, it is important for now ?
Total trip distance - done using @digitalentity functions

Things to do is move this to actual 1.6 tree - I've made test on 1.5.1 but probably there is not so many differences in 1.6 OSD, and add some new signs to all fonts used in GUI. Like I said before on slack, this should be done this weekend.

martinbudden, 2017-02-10:
@marbalon , great. I'd make the PR without the Next Waypoint Direction - that can be added in later.

digitalentity, 2017-02-10:
@marbalon Next Waypoint Direction is not important - it can me added later if necessary. At the moment I'd like to give users current coordinates, direction to home and some statistics about distance and trip.

Schalonsus, 2017-02-10:
Just a question regarding flightmodes...
In my OSD (microMinim with MWOSD) when having LaunchMode activated, it shows me the flightmode it goes after launch is finished.
Also Passthrough is shown as Gyro and Acro is also shown as Gyro... is this a problem with iNav code or with MWOSD code?

sppnk, 2017-02-10:
As @Schalonsus said beforre, I guess an indication on OSD of instantaneous Wh/km and averaged Wh/km (at end of flight) will be a good thing for people trying to find best efficiency, playing with AUW, propellers, motors, batteries, and so on. MWOSD shows this value, (although many of us use the old "charged mAh vs flight km" method).

And perhaps later this value could be used for automated efficiency flights. A nice thing to play with....

oleost, 2017-02-10:
Mwosd fault I believe. It needs to be aligned with https://github.com/iNavFlight/inav/blob/master/src/main/fc/fc_msp.c#L112 when you define that you are using iNav.

https://github.com/ShikOfTheRa/scarab-osd/blob/4232212e849c269aeb60e05af59e3cd57214a26f/MW_OSD/Serial.ino

digitalentity, 2017-02-10:
@sppnk let's look into more OSD features once we have basics working.

sppnk, 2017-02-10:
sure @digitalentity , this was only an idea for future. :)

Schalonsus, 2017-02-10:
@sppnk would be better to have this Wh/Km in flight. So you can manage your throttle for most efficiency long range flight, but i agree with DE this would be something for future releases

digitalentity, 2017-02-14:
General support for nav-specific OSD is there. Closing this one.

closed 2017-02-14 by digitalentity