iNavFlight/inav #931: no baro and mag detected on SPREVO on  eeaa20a

bk79, 2016-12-14:
this is the status and bootlog
# status
System Uptime: 32 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.01
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, SONAR=NONE, GPS=NONE
Stack size: 2048, Stack address: 0x10002000
Cycle Time: 2010, I2C Errors: 0, config size: 2032

# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 501: 19   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 501: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (9, 9, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (10, 10, 0, 2)
 501:  2          PWM_INIT_DONE 
 511:  3       EXTRA_BOOT_DELAY 
1047:  9         GYRO_DETECTION  (7, 0, 0, 0)
1639: 10          ACC_DETECTION  (8, 0, 0, 0)
1639: 11         BARO_DETECTION  (0, 0, 0, 0)
1639: 12          MAG_DETECTION  (0, 0, 0, 0)
1639: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
1639:  4       SENSOR_INIT_DONE 
2340:  7    TELEMETRY_INIT_DONE 
3342:  8           SYSTEM_READY

stronnag, 2016-12-14:
Expected behaviour. For 1.5 you need to explicitly set the hardware.

bk79, 2016-12-14:
Not exactly user friendly...i don't think less expert people to mess with
cli to find the correct hw

stronnag, 2016-12-14:
They can find it from the wiki and enter the values in the updated configurator. Maybe even that can get automated from the FC identification. Given the myriad of problems with auto-detection, this will at least be reliable. Care to close this non-issue?

bk79, 2016-12-14:
Close it

closed 2016-12-14 by oleost

digitalentity, 2016-12-15:
Autodetection caused all kinds of issues, it even may misdetect MS5611 for BMP085 (or vice versa I can't remember). It may also cause problems if more than one chip is available on the bus - i.e. it may result in onboard compass being used instead of external one. 

While new solution is a bit less user-friendly it's considerably more robust.

bk79, 2016-12-15:
It would be nice to have the possibility to configure it from the
configurator l

digitalentity, 2016-12-15:
That's the idea!

DzikuVx, 2016-12-15:
@bk79 Configurator will follow shortly