iNavFlight/inav #938: Profile change activates Failsafe

DzikuVx, 2016-12-16:
In some cases, changing profile 1 to profile 2 causes Failsafe state to activate when receiver is not connected

digitalentity, 2016-12-17:
Hm... I'll have a look

digitalentity, 2016-12-19:
@DzikuVx I'm unable to reproduce the issue. What board did you use? Dump will also help.

DzikuVx, 2016-12-20:
I will try to write more datails and reproduce again. 
Setup is: Omibnus F4, X4RSB, SBUS

oleost, 2016-12-20:
Tried replicating, no sucess. 

Setup AirbotF4. Frsky XSR connected with SBUS.
Changed rateprofile trough Aux1 changing between 3 different profiles. (Setiup using adjustment tab )

oleost, 2016-12-20:
Tried replicating, no sucess. 

Setup AirbotF4. Frsky XSR connected with SBUS.
Changed rateprofile trough Aux1 changing between 3 different profiles. (Setiup using adjustment tab )

Also disconnected the receiver and change profile with rateprofile command in CLI, did not see the failsafe symbol lit up there either.

DzikuVx, 2016-12-20:
It was strange because if I remember correctly, it was engaging failsafe only from profile 1 to 2. Not from 2 to 1

closed 2017-01-30 by digitalentity