iNavFlight/inav #949: RFC: GIT_SHORT_REVISION_LENGTH is 7, git now gives 8 digit lengths

stronnag, 2016-12-18:
This does not appear to cause any problems for iNav (only to buggy black box replay tools:). 
As of recently, `git log -1 --format="%h"` returns 8 characters, as seen in bbox logs.

Please close this if it's of no consequence.

````
$ git --no-pager log -1 --format="%h"
0eb6e821
````

````
$ grep -r --include \*.h GIT_SHORT_REVISION_LENGTH .
./src/main/build/version.h:#define GIT_SHORT_REVISION_LENGTH   7 // lower case hexadecimal digits.
````
````
# INAV/NAZE 1.5.0 Dec 18 2016 / 18:17:22 (0eb6e821)
````
````
$ git --version
git version 2.11.0
````

stronnag, 2016-12-18:
This means we get different results in BBlogs (8 digits) and via MSP BUILD_INFO (7 digits).

digitalentity, 2016-12-19:
This probably requires fixing.

stronnag, 2016-12-19:
Here's where it happened:
````
len    log file                        Short ID
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
7      ./2016-12-11_tri/LOG00008.TXT   ed7f767
8      ./2016-12-16_tri/LOG00001.TXT   cc121438
````

digitalentity, 2016-12-23:
Fixed.

closed 2016-12-23 by digitalentity