iNavFlight/inav #959: EB185 Drone by Eachine LED Show as 0 remaining

tonyintn, 2016-12-21:
I am not sure if this has been address, but I have the EB185 Drone by eachine it will not allow me to program the read addressable leds, it always show as 0 leds remaining even though there are 6 in the rear of the quad.
I flash CleanFlight and BetaFlight and it show as 32 available and allows me to program them; Only when I flash with Inav with is prefer will it now allow the leds to program.
I also tried to use CLI interface and manually push and save led configuration but it will not save.
Is there a fix for this one? or has it been addressed?
Thanks in advance.

DzikuVx, 2016-12-21:
Due to flash size limitation INAV does not supports LED on Naze32 target
https://github.com/iNavFlight/inav/wiki/Hardware-and-feature-support-map

closed 2016-12-21 by DzikuVx

tonyintn, 2016-12-22:
I noticed that in that like provided it says you can compile your own version to maybe for example remove something and add item. 
Is it feasible to maybe have a compiled without blackbox and add the LED feature.  If I am asking a stupid question is apologize.  I am just trying to think of a way to get the led feature in my version.  If you can recommend someone that does this I do not mind donating to have it done.
Thanks again.

martinbudden, 2016-12-22:
@tonyintn , it would probably be possible to do a build without blackbox and with LED.

Can I ask why you want to use iNav rather than betaflight? The EB185 is a "racing drone" and has no GPS, so why not use betaflight?

stronnag, 2016-12-22:
It is entirely feasible, and removing blackbox would probably work for you.
The problem is the huge number of combinations, and the impossibility to satisfy everyone.

http://www.seyrsnys.myzen.co.uk/naze_leds_nobb/inav_1.5.0_NAZE.hex.gz

tonyintn, 2016-12-22:
I have added gps to if; a noob I know but I wanted to to have it :) it's has a ublox 8 mini works great all is good just the Leds were not working I do understand why now but I was just trying to get them going.

tonyintn, 2016-12-22:
Fat fingers sorry :) I am watching a how to vid now so I can learn how to do it. 
Already lost one drone from a flyaway tryng limit my chance of that again by adding gps.

tonyintn, 2016-12-24:
Thank you for the file, I am attempting to create it myself using Windows 7 following this video https://www.youtube.com/watch?v=n3Z1fOQJAg8&feature=youtu.be
It compiles but gives tons or warnings similar to the below I am getting the latest inav and syncing with the latest inav 1.2.

%% encoding.c
%% filter.c
./src/main/common/filter.c: In function 'biquadFilterInit':
./src/main/common/filter.c:121:21: warning: 'b2' may be used uninitialized in this function [-Wmaybe-uninitialized]
     filter->b2 = b2 / a0;
                     ^
./src/main/common/filter.c:120:21: warning: 'b1' may be used uninitialized in this function [-Wmaybe-uninitialized]
     filter->b1 = b1 / a0;
                     ^
./src/main/common/filter.c:119:21: warning: 'b0' may be used uninitialized in this function [-Wmaybe-uninitialized]
     filter->b0 = b0 / a0;
                     ^
%% maths.c

stronnag, 2016-12-24:
You will not get help using an ancient version like 1.2. iNav 1.4 / 1.5 compile cleanly with a modern compiler.
Building the capability you want is very simple, in `./src/main/target/NAZE/target.h`

* remove the comment for `#define LED_STRIP`
* add `#undef BLACKBOX`

tonyintn, 2016-12-24:
sorry i mean i pulled  1.5-RC2 and arm tool chain 4.8-2014-q2.

tonyintn, 2016-12-28:
think I figured it out, I was pulling the  4.8-2014-q2 version but when I pull the most current 6.2 2016q4 it build without any warnings.