iNavFlight/inav #970: Fixed Wing Pterm attenuation when sticks are centered

DzikuVx, 2016-12-26:
Current implementation on PID on FixedWing tends to very active control surfaces behavior even during level flight. This is very visible on small flying wings. 

Airplane has enough "build in" stabilization not to require such active Pterm.

Idea is to attenuate Pterm when requested rotation rate is `0` and then scale it lineary to 100% at given rotation threshold. 

For example: Pterm is attenuated to 40% at `0` requested rotation and then scaled up to 100% at 100dps.

Due to asymmetrical FW characteristics, it might be wise to have `0` attenuation and threshold separate for roll and pitch

digitalentity, 2016-12-26:
Makes sense. Another extension - implement feed-forward control which will connect requested angular rate directly to control surfaces bypassing gyroscope and rate error.

digitalentity, 2016-12-26:
Until https://github.com/iNavFlight/inav/issues/969 gets implemented we can hijack D for feed-forward gain (there's no use for D on airplane anyway)

DzikuVx, 2016-12-26:
I'd rather not use D. Lets keep it simple and obvious. D is D and let it
stay D

digitalentity, 2016-12-26:
@DzikuVx I'm going to do some experiments on this in the following days. Will try different scaling for P-gain (different approaches to TPA), also P-gain scheduling according to rate setpoint and feed-forward control.

DzikuVx, 2016-12-26:
OK, looking forward for the results

digitalentity, 2016-12-29:
@DzikuVx weather is not friendly here - sub-zero temperatures down to 30 deg + wind. If you have the time - please have a look at https://github.com/iNavFlight/inav/pull/995. It's built around feed-forward idea (as an experimental code).

DzikuVx, 2016-12-29:
Ok, i might have a chance to fly it tomorrow. Will take laptop to the field

closed 2017-01-28 by digitalentity