iNavFlight/inav #974: Add an option to reset altitude to zero on each ARM

digitalentity, 2016-12-27:
As discussed here - https://github.com/iNavFlight/inav/issues/546#issuecomment-269243222

carlonb, 2017-02-03:
Hey DE, 
Are you considering to add this feature in final version 1.6 ?
Thanks in advance
Carlo

carlonb, 2017-02-04:
Thanks digitalentity, 
I saw it now, you are working very hard on iNav.....
I'm waiting for the final release.
Bye
Carlo

closed 2017-02-05 by digitalentity

carlonb, 2017-03-18:
Hi DE,
Test right now on my desk the new inav 1.6.1 with configurator 1.6.3, outside there is a strong wind :(
Here the cli diff:
`# diff

# version
# INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature GPS
feature RX_PARALLEL_PWM
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 2 115200 57600 0 115200

# led
led 0 0,0::G:0
led 1 1,0::G:0
led 2 2,0::FW:0
led 3 3,0::FW:0
led 4 4,0::FW:0
led 5 5,0::FW:0
led 6 6,0::AW:0
led 7 7,0::AW:0
led 8 8,0::AW:0
led 9 9,0::AW:0
led 10 10,0::FW:0
led 11 11,0::FW:0
led 12 12,0::FW:0
led 13 13,0::FW:0
led 14 14,0::G:0
led 15 15,0::G:0

# color

# mode_color

# aux
aux 0 1 0 900 2100
aux 1 4 0 900 2100
aux 2 3 0 1400 2100
aux 3 11 3 1300 2100
aux 4 15 3 1300 2100

# adjrange

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 55
set gyro_notch1_hz = 144
set gyro_notch1_cutoff = 90
set gyro_notch2_hz = 72
set gyro_notch2_cutoff = 50
set acc_hardware = MPU6050
set acczero_x = 25
set acczero_y = 71
set acczero_z = -805
set accgain_x = 4065
set accgain_y = 4060
set accgain_z = 4051
set mag_hardware = HMC5883
set magzero_x = 204
set magzero_y = -234
set magzero_z = 146
set mag_hold_rate_limit = 30
set mag_calibration_time = 60
set baro_hardware = MS5611
set blackbox_rate_denom = 2
set failsafe_procedure = RTH
set vbat_scale = 107
set vbat_warning_cell_voltage = 34
set gps_sbas_mode = AUTO
set inav_use_gps_velned = NONE
set inav_reset_altitude = EACH_ARM
set nav_disarm_on_landing = ON
set nav_extra_arming_safety = OFF
set nav_max_climb_rate = 400
set nav_manual_speed = 400
set nav_land_slowdown_maxalt = 1000
set nav_rth_alt_mode = CURRENT
set nav_mc_bank_angle = 35
set async_mode = GYRO

# profile
profile 1

set dterm_lpf_hz = 30
set dterm_notch_hz = 108
set dterm_notch_cutoff = 72
set rate_accel_limit_roll_pitch = 360
set rate_accel_limit_yaw = 90
set roll_rate = 18
set pitch_rate = 18
set yaw_rate = 9

# 
`
so setting "set inav_reset_altitude = EACH_ARM" the altitude drift is never zeroed at each arm. What about this ? May be a bug or something that I've not understood ?
All other param and setup seems good.

Thanks and bye
Carlo

carlonb, 2017-03-18:
One more info,
Connected with configurator in sensors tab I see Baro data changing quickly (this is normal), and after the first arming, Altitude is changing smoothly (also this is OK), but there is an amount of drift, so if I disarm/arm, the drifted altitude is never zeroed.
Bye

digitalentity, 2017-03-18:
Thanks for reporting. I'll check!