iNavFlight/inav #976: Allow fully autonomous operation

digitalentity, 2016-12-27:
Allow a waypoint mission to be executed in a fully autonomous way - arming by telemetry command, launch by LAUNCH assistant, automatic landing via parachute or normal landing glissade.

b14ckyy, 2018-10-22:
+1 vote for this feature.... 

lost my plane yesterday because disabled Failsafe to do an out of range mission. Forgot that FS is disabled and in cruise mode lost connection…. 

Yes... my fault... I was just dump to risk that…. but would really appreciate this feature in the future

peteoz, 2019-05-22:
another +1 vote for this feature.

may I also suggest to please re-open request #3937 , I understand it was marked as duplicate of this request (#976) but that's not really correct :
#976 is more complicated implementation of fully autonomous operation from autolaunch to landing, including parachute option
whereas
#3937 is a simpler request to enable us to continue WP Mission in case of radio link loss (for example as an alternative option in CLI settings), request #3937 was well written, down to the point and brief

request #976 has been inactive for 2 and half years, maybe it's too complex to implement, I would suggest to implement request #3937 first and later that functionality can be extended to full autonomous operation as requested by #976

b14ckyy, 2022-03-23:
can't this be closed now? Kooks like a request corpse of a already implemented feature.