iNavFlight/inav #977: New FAILSAFE Stage 2 procedure

digitalentity, 2016-12-27:
For better user-friendliness we need a smarter failsafe procedure that will act depending on current flight mode and flight conditions.

If UAV is flown manually - RTH (or land if no GPS is available)
If UAV is executing autonomous mission - continue mission

Voltage failsafe/geo-fencing can be part of this "smart" failsafe.

What do you think?

stronnag, 2016-12-27:
I like this proposal. geo-fencing welcome. Note that there is extant MSP_(SET_)_NAVCONFIG  message permitting the setting / getting of (inter alia) geo-fence radius, however there may be too much legacy junk there and we'd be better to do our own control message.

oleost, 2016-12-27:
"If UAV is executing autonomous mission - continue mission" is basically #770 

Please make it user configurable. Like for me personally I would never let the aircraft do WP mission by itself.

There are high risk running WP mission without TX control and this should be a conscious choice.

digitalentity, 2016-12-28:
@oleost yes, it's #770. I just want this to be a more flexible failsafe mechanism. I might as well make a failsafe mode matrix (user-configurable) where you'll be able to chose different procedure for different flight mode.

DzikuVx, 2017-02-02:
@digitalentity I think this should be closed now. Am I right?

digitalentity, 2017-02-02:
@DzikuVx not yet. We still can't have one setup that will initiate failsafe RTH when flying manually and do nothing when on WP mission. I'll just reschedule it for 1.7 since most of the implementation is already there.

oleost, 2017-03-28:
#1475

closed 2018-05-27 by stale[bot]