iNavFlight/inav #980: bidirectional mavlink telemetry

anemostec, 2016-12-27:
in order to be able to use other gcs such as tower or mission planner I propose to add the mavlink bidirectional link support to inav.
In order to do that, we will fetch the relatvie code from ardupilot and bring it to inav. Will that be of any interest?
what pitfalls do you see in this dev approach?

oleost, 2016-12-27:
How would this cooperate with MSP?

And is there wiser to have an layer translating mavlink to MSP somehow ?

anemostec, 2016-12-27:
the input mavlink telemetry will be used mainly for navigation purpose, thus those input messages will call the exact same navigation function as the MSP messages. MSP is just a layer in between, I don't see the benefit to call them rather than the final destination functions

stronnag, 2016-12-27:
I think there is a danger it would make the message parsing layer much more fragile than it is now. As Tower is open source, why not fix it it allow unsolicited telemetry rather than adding complexity to iNav?

anemostec, 2016-12-27:
tower is working actually with inav
the waypoints support, flight mode changing is not since the inav mavlink telemetry is just an output one, no input supported yet

digitalentity, 2016-12-28:
Flight mode changing won't be possible as it's only driven by RC RX driver. Uploading a mission via MAVLink would be nice - you are welcome to submit a PR for bi-directional telemetry.

anemostec, 2016-12-28:
ok for the bi-directional telemetry, but I still don't understand why we 
could not call the function used by the RC RX driver to change the 
flight mode

digitalentity, 2016-12-28:
Because next update from RC RX will change flight mode back according to transmitter switches.

anemostec, 2016-12-28:
oh ok, got you

but we could put a global variable to tell RC RX to not change flight 
mode back if flight mode was already changed by telemetry unless the 
user willingly change the RC RX switch position

magictex, 2017-03-10:
Is someone working on this issue ?
Simple question because I'm also interested and can maybe help to port Mavlink incoming messages to iNav.
Thanks

martinbudden, 2017-03-10:
@magictex , no one is actively working on this at the moment. But if you wish to submit a PR it would be most welcome.

magictex, 2017-03-10:
@martinbudden Thanks for the answer. Will check with a friend of mine and work on it then.

digitalentity, 2017-03-11:
@magictex I have this issue in my schedule after merging in some sonar-related code.

magictex, 2017-03-11:
@digitalentity Thanks for the info. Will be glad to test it when you'll release it ! :-)

digitalentity, 2017-03-26:
Closing - let's continue in relevant PR #1461

closed 2017-03-26 by digitalentity