iNavFlight/inav #982: Make "mag" and "heading lock" work with rudderless airplanes / fixedwing

oleost, 2016-12-28:
Make "mag" and "heading lock" work with rudderless airplanes / fixedwing

digitalentity, 2016-12-28:
Putting this into 1.7 milestone. Not sure we can create something working in 1.6

marbalon, 2017-01-29:
I think it is quite easy, just like someone told before, set the WP far away using current bearing. Maybe we can put this in 1.6 ? ;-) After this week I will try to focus on testing 1.6 on wing so can test this too. It is really useful feature for planes.

digitalentity, 2017-02-01:
Yes, it's a useful feature, but we have a handful of features in 1.6 already. If we won't do feature freze we'll never release it :smile:

digitalentity, 2017-04-04:
Oops, totally forgot about this. Will have to move to 1.8

closed 2017-05-24 by digitalentity