iNavFlight/inav #985: It's possible to overwrite HOME position in flight

digitalentity, 2016-12-28:
We don't protect WP#0 (HOME) from being overwritten using MSP_SET_WP. We don't even have it protected by `NAV_GCS` mode. In presence of noise on MSP connection we can have a malformed MSP_SET_WP command changing our HOME which will cause RTH to fly in random direction.

Possible solutions:

1. Have WP#0 protectected by `NAV GCS` mode
2. Remove the possibility to modify HOME position by telemetry (why would one do that anyway)?

I'm in favor of the second option.

oleost, 2016-12-28:
I believe DJI has the ability to have home position updated to position of controller ( phone GPS )

This is example if on a moving boat, walking / driving while filming an road etc

martinbudden, 2016-12-28:
As I understand it, that's how "follow me" works - by continually updating the home position. Maybe that's not the best way to do it, but I believe there are people who use it in that manner and so it shouldn't be removed until there is a better alternative.

digitalentity, 2016-12-28:
@martinbudden "follow me" works by modifying WP#255 (current PosHold position), home is not touched in this mode.

wilco1967, 2016-12-28:
Great to hear I might have accidently contributed finding this issue... ;-)
I think, home should never be modified by anything once armed.  

Another suggestion is don't allow arming if the first waypoint is more than say a 100 meters from 'home', or when the total WP mission (and back home if it doesn't end near 'home') is more than say 5 km (preferably adjustable by some configuration setting).

Changing the WP list while flying might be useful for some, but if possible, I would personally prefer this to be only enabled by a configuration setting (not by default).
And if doing so, some check that the new waypoint list is not beyond the capabilities of the remaining battery / flight time / or similar... a failure of such a check should cause a RTH I would suggest...

'geo fencing' would be a nice addition, but not too important if above protections are in place...
 (perhaps above could be considered some kind of geo fencing).

Thanks a lot !!!!
still love this great firmware...

oleost, 2016-12-28:
Is there isn't a way to simply bulletproof waypoint changing?

Like example followed by new home coordinates an valid hash sum of the coordinates must also be sent by the groundstation?

Also it could be additional configurable to be allowed to change homepoint.

martinbudden, 2016-12-28:
There are a number of good suggestions here, to enumerate:
1. Have specific configuration setting to allow home waypoint to be changed after arming.
2. Have specific configuration setting to allow other waypoints to be changed after arming.
3. Do not allow home waypoint to change by more than say 100 meters.
4. Add checksum or CRC to waypoint change command to guard against inadvertent changing.

stronnag, 2016-12-28:
I don't see how we have any evidence to show that the home position was changed by the CGS. From the narrative so far, I do wonder whether there were some other memory integrity issues with the hardware "After I changed the battery, ... I noticed the old route was still partly in memory, but with a few random stray waypoints added". It is also the case that the MSP checksum is weak (compared to either of Mavlink or the UBlox protocol), so it's possible that a couple of single bit errors could generate a corrupt WP #0. Validation of a mission before take off is always advisable; mwp does this and if ezgui doesn't, then a feature request to Bart may be in order. So some thoughts:
* Do not allow the CGS to set WP#0 (unless a feature that allows it is explicitly set);
* Do not allow any WP to be altered while armed other than WP#255 when mode GCS is enabled

wilco1967, 2016-12-28:
@stronnag 
Correct, we don't have any hard evidence, just a guess.
But still, I think home waypoint should not be easily changed.... even if it may or may not have caused this fly-away, the potential shouldn't be there in the first place.
I had problems with configuration corruption using the same 433 UART radio link, on Multiwii before.... From what I understood, it seems the uart radio software is not a transparent datalink, but only allows MSP commands it recognizes.... so if the radio link is weak, it could possibly 'think' it hears a valid MSP command, and then sent that to the FC... This I think is much more likely than a pure random radio signal.... 
Also the control signals from Uart radio (which I didn't use now) are prone to corruption... 
This UART radioi firmware never saw much use as far as I know, and Alexinparis didn't develop it further.

I've tried to use OpenLRS and Ultimate LRS software, but never managed that to work properly.  

This copter hardware has always been very reliable. Earlier I had some troubles with unexpected behaviour, but after I changed the FC a few months ago, it has never had any strange issues. it always flew perfectly. Never lost GPS , I2C, whatever.
Sure, something hardware could have failed, but we also have no evidence for hardware failure. And a hardware failure would more likely cause the copter to come down on the spot (it isn't there... i checked ;-).

Not sure if the old route showing up in EZgui is because EZGui read it from the FC after it rebooted,  or it existed only in EZGui memory I don't know.... Or it picked it up from stray signals while the FC was down.
Also worth mentioning: EZgui often remains normally connected via bluetooth, even though the FC rebooted. I cannot remember if I needed to reconnect bluetooth this time, but often I don't have to, if the FC is down only for a short while... 

If required, I can upload the Frsky and EZgui logfiles somewhere. Let me know if it helps.
There are some strange values occasionally in the EZGui log, which seem like data link corruption.. these are not present in the Frsky telemetry (values in Frsky are either normal, or missing / not updated).

wilco1967, 2016-12-28:
some additional information:

my datalink is as follows
2x 100 mw Orange open LRS RX modules. one operating as Tx, the other as Rx.
mounted in the module bay of my Taranis radio.
Taranis PPM out signal is connected to the 433 module, and is sent out. but iNav is configured to PPM signal from the Frsky XSR transmitter so I assume it ignores the control signals it receives via MWP protocol. This MIGHT be a hint.... before I tried to get it working, but previous iNav version (1.4) apparently wasn't working... perhaps the 'fix' in 1.5RC2 may cause an issue with RTH when MWP control signals are present ???? Just a wild guess again.

EZgui connect via a bluetooth adapter to the 433 module in the back of the Taranis (taranis provides power through a 5V Ubec).

Frsky telemetry is S-port via UART-2
433 telemetry via UART-1 / MWP protocol
UART-3 = GPS
No OSD on this flight.

I will not be able to test anything with 433, as I also lost that part, and I didn't order a replacement (yet).
I really have the feeling, that 433 module (or more accurately, the firmware) may possibly have contributed.

digitalentity, 2016-12-29:
For the time being I'll only allow to modify WP#0 (home) and WP#255 (hold) if `NAV GCS` mode is activated from radio. This will somehow increase safety until we implement more checks.

digitalentity, 2016-12-29:
I'll keep this open and move to 1.6 so we can discuss further and make better protection

Riffer, 2017-01-27:
Is the method of setting WP#0 the same as RTH set by mode switch?

If not - is the RTH point protected anyway? Usually I connect the battery to the fc while the model (fixed wing) is on the table. After having all checks I then move the model onto the flight place and set the RTH by mode switch to be there. Is this the correct way further on?

digitalentity, 2017-01-29:
No, this PR affects only changing "HOME" via telemetry. You approach is safe - home is locked after you arm the machine - before that home is constantly adjusted automatically to current localtion.

closed 2017-01-31 by digitalentity