iNavFlight/inav #989: RX_Parallel_PWM AETR1234 Channel Map Broken in 1.5RC3 for Naze

dbechth, 2016-12-29:
The channel mapping appears to be broken in 1.5RC3 for the Naze hardware. Flashing 1.5RC3 with full chip erase and then setting receiver mode to RX_Parallel_PWM causes ThrottleStick to come in as Aux1, YawStick comes in as Aux2, Aux1Switch to come in as Aux3, and Aux2Switch to come in as Aux4. RollStick and PitchStick do not seem to come in anywhere. I have switched back to 1.4 stable as a sanity check and confirmed that everything is working as expected.  For further investigation I can confirm that 1.5RC1 has the same behavior as RC3.

digitalentity, 2016-12-29:
Please upload the full dump of your configuration.

digitalentity, 2016-12-29:
Also, please test if the issue is solved in this hex.
[inav_1.5.0_NAZE.hex.gz](https://github.com/iNavFlight/inav/files/676581/inav_1.5.0_NAZE.hex.gz)

dbechth, 2016-12-29:
The board is a NAZE32 Rev6. See full 1.4 dump below. I have tried loading this to 1.5RC3 with the result described above. I will try the .hex you suggested and report back.

`# dump

# version
# INAV/NAZE 1.4.0 Nov 28 2016 / 10:55:47 (6bf7a91)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature VBAT
feature GPS
feature FAILSAFE
feature RX_PARALLEL_PWM
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
[... 278 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 100
set rc_yaw_expo = 0
set thr_mid = 43
set thr_expo = 50
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 25
set tpa_rate = 0
set tpa_breakpoint = 1500

# `

dbechth, 2016-12-29:
I loaded the .hex and my cli dump and everything looks as expected. Its too dark to fly right now but ill try to fly tomorrow after work. Thanks for the ultra fast response! Please let me know if there is anything else I can test.

digitalentity, 2016-12-29:
Great, thanks for testing!

digitalentity, 2016-12-29:
Fixed

closed 2016-12-29 by digitalentity