iNavFlight/inav #993: ArduSoar support - flying all day with sail plane

Schalonsus, 2016-12-29:
Some guy is porting ArduSoar to Patricks Basflight fork.
Would be nice to have this feature in iNav too.
https://github.com/brm007/baseflight
https://github.com/samuelctabor/ardupilot/wiki/Ardusoar:-Soaring-fork-of-ArduPlane

Of course this is a long term process i think, but having a look at it should be worth.

digitalentity, 2016-12-29:
I would really like to understand math behind the code. Any idea if there is any paper on this?

Schalonsus, 2016-12-29:
This is all i have found for this topic...
ArduSoar:
https://github.com/pbraswell/ArduSoar
http://diydrones.com/forum/topics/autonomous-soaring?xg_source=activity
http://diydrones.com/profiles/blogs/ardusoar-cross-country-x-plane-simulation

ThermoPilot:
https://github.com/jlnaudin/x-drone/wiki/x-drone:-FX-61-Phantom-flying-wing-THERMAL-SCAN-mode-test-flight-with-the-ThermoPilot-v7.2
http://diydrones.com/profiles/blogs/thermopilot-project-a-thermal-hunter-glider-drone

martinbudden, 2017-01-27:
@digitalentity , I've just had a look at the code and it's really simple. In soaring mode it just does the following:

1) Detect if aircraft is in a thermal (true if climb rate is higher than a set threshold).
a) if in a thermal, then go into circling flight pattern.
b) if not in a thermal, return to normal flight pattern.

That's it.

martinbudden, 2017-01-27:
Some of the other implementations, when out of thermal mode, go into a "thermal scan" flight mode - basically fly around looking for a thermal.

Landflieger, 2018-05-13:
Season has started, time for doing some flights with my glider. Any progress in this ?

digitalentity, 2018-05-13:
@Landflieger no progress. Developer resources are limited, but you are welcome to help implementing this feature.

closed 2018-07-27 by stale[bot]