iNavFlight/inav #994: Fixedwing: Enable landing after returned from failsafe initated RTH

oleost, 2016-12-29:
After returning home when RTH has been initiated by failsafe it should start descent and then cut throttle.

Suggestion:

1. After returned home continue loiter and start descending using althold controller, at and set rate.
2. When altitude is reached height of example 30 meters cut throttle. ( Go to minimum command ) ( Dont disarm so if pilot can regain control he doesnt need to re-arm )
3. Continue descent.

Optional added features.

Some kind of stall detection that dont relay on increasing throttle.
Example if GPS speed (or pitot) is below speed guard instead of increasing throttle increase descent ?

oleost, 2016-12-29:
And I know PatrikE firmware have something like this in his firmware. That have been proved succesfull

digitalentity, 2016-12-29:
Good, we need to look into it.

oleost, 2016-12-29:
This is what PatrikE replied to me awhile ago:

Hello,
In Failsafe RTH with GPS.
When the plane arrived home close engine and circle down.
Stages
1 / Enable posHold / Lotier and change the target Altitude to 5m and disarm.
2 / Change target to Altitude 0m.

without GPS
1 / Run engine at Failsafe throttle ~ 20 sec and apply ~ 5 degrees left roll to do a blind circle back to catch the signal.
2 / Disarm, continue circle and apply ~ 5 degree dive to stop runaway.

Options to step 2 if it is windy is to take down the plane really quickly it can drift quite far at high altitude.
If you add it up in the tough role of> 45 degrees and little dive then it will come down relatively quickly.

Happy Holidays..
Patrik

oleost, 2016-12-29:
https://drive.google.com/open?id=0B-yHzNtOXxbEYnNkQjJ3V2t1eDg

This is his firmware, the source.

digitalentity, 2017-01-30:
Let's make this more general

PR #1236 (2017-02-05): RTH improvements
- [x] Option to ignore GPS loss (don't execute emergency landing) during initial stage of RTH. CLI: `nav_rth_climb_ignore_emerg`
- [x] Option to enable/disable landing at last stage of RTH (all aircrafts). CLI: `nav_rth_allow_landing`

Closes #994 and #964
closed 2017-02-05 by digitalentity