iNavFlight/inav #1000: Altitude hold with MSP

liutairan, 2016-12-30:
I am using Naze32 rev6 10DOF with nav 1.5RC3.

What I am trying to do is click on one button on my computer, then my quad can take off, get to the altitude I set, then keep that altitude for several seconds, then landing.

I am using XBee to do all of this. Now I am able to use my GUI to control it take off, but how to let to go to the altitude I set then keep that altitude autonomously?

Which MSP command or code can I use to get this?

Thanks!

liutairan, 2016-12-30:
I guess this is part of waypoint navigation or some more low level function, right? But I just don't know what that MSP command is.

liutairan, 2016-12-30:
By the way, I want it hold a very low altitude with sonar. The distance to the ground can be 50cm or 80cm or some value like this.

oleost, 2016-12-30:
https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages

https://github.com/iNavFlight/inav/wiki/8.-Navigation-modes#gcs_nav---ground-control-station

liutairan, 2016-12-30:
@oleost 
Thanks a lot. These pages really changed a lot since last time I read it. Do you or your team have a plan to develop a cross-platform GCS? I really wish there is one which keep pace with inav updates because if it is related to navigation, GCS is so important. 

I ever tried the mwp two months but met some problems with installation.

I will check the updated pages you just mentioned. Thanks again!

stronnag, 2016-12-30:
You can (almost) do what you want with either mwp or ezgui.

* Enable autoland on RTH
* Create / upload a mission with a single POSHOLD WP at the desired altitude

Then:

* Arm the vehicle
* Engage WP mode

The vehicle will take off, fly to the PH WP, and hold there at the set altitude. When you're done

* Engage RTH

Unfortunately, iNav does not (yet, hint) support the MW/MSP Timed POSHOLD WP; if it did you could do want you without manually engaging RTH; just have mission with a timed PH WP and RTH.

liutairan, 2016-12-30:
@stronnag Yeah, thanks again! I replied you under your repository, I got it work. Thang you so much.

I checked the GCS page and found your mwptools and some other tools. As I don't have an Android device so I just moved to mwp.

closed 2016-12-30 by liutairan