iNavFlight/inav #1003: Roll & Pitch Reversed

Eric032990, 2016-12-30:
Board: SPRACINGF3 Deluxe
iNav firmware: 1.5.1
Configuration: Ublox M8N GPS hooked up to UART2, Using FC compass. FS-IA6B connected to UART3

Issue explained:  when I connect my FC to iNav the direction of the boar is correct. I changed YAW degree to 270 since I have the USB coming out the side and not the front or back. The problem Im having is that with the quadcopter is sitting flat and facing forward away from me if I lift the back side the front goes down but on iNav it shows the nose going up then if I lift the front the rear goes up. If I lift the left side then the right side goes up. Basically everything is backwards.﻿

CLI DUMP:

`# dump

# version
# INAV/SPRACINGF3 1.4.0 Nov 28 2016 / 10:59:44 (6bf7a91)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature VBAT
feature RX_SERIAL
feature GPS
feature FAILSAFE
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
[... 387 lines cut ...]
set max_angle_inclination_pit = 300
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500`

DzikuVx, 2016-12-31:
instead of `set align_board_yaw = -900` try 2700 and check again if it is correct value

digitalentity, 2017-01-04:
@Eric032990 I suppose you should use `set align_board_yaw = 900` - you basically rotated the board in wrong direction in Configurator.

Schalonsus, 2017-01-09:
Have the same problem with a F1 target. Also using set align_board_yaw = 2700
I will try if setting it to 900 is the solution asap.

Schalonsus, 2017-01-09:
Tested "900" and the axes are going inverted.
With "-900" the axes move right but problem persists.
I'm using this board http://www.banggood.com/DALRC-NAZE32-REV6-MPU6050-32-bit-6-DOF10-DOF-Flight-Controller-for-Multicopter-p-1020774.html?rmmds=myorder
Maybe they put a chip the wrong way on the board?

digitalentity, 2017-01-10:
@Schalonsus so what's the problem exactly?

270 deg is **not** 90 deg - those are direct opposites and both roll/pitch will be inverted between the two.

Schalonsus, 2017-01-10:
The problem is: if i change roll value in configurator it changes pitch axis and vice versa
But in setup tab it moves all right. Just the values are interchanged.

digitalentity, 2017-01-10:
INAV does read sensors in board frame of reference.

Board alignment are relative to board, not to the frame. So once you rotate the board 90deg roll/pitch alignment are switched **relative to frame**. 

This about this as of rotating the frame around the board.

Schalonsus, 2017-01-10:
Ok, so this is no problem or bug.
Thank you.

DzikuVx, 2017-01-11:
OK, since problem solved, I'm closing this one

closed 2017-01-11 by DzikuVx