iNavFlight/inav #999: Level flight problem

liutairan, 2016-12-30:
I am using the Naze32 rev6 10DOF with nav 1.5RC3. I am currently facing a level problem that when I am trying to trim the acc sensor, if I change board and sensor alignment roll to -0.9, then my quad will go slowly to the left after taking off, and it I set it to -1.0, it will go slowly to the right after taking off. I found that I cannot set the value to -0.95 or something has one more digit. So how I can do with this if I want it just keep the position?

I am still doing indoor experiments and haven't tried GPS yet. But I think the problem is about the level.

Anyone has any idea about this?

Thanks!

digitalentity, 2016-12-31:
0.05 deg won't do you any good. Slightest airflow will cause more drift than such small misalignment. If you want your quad to be able to hold position indoors, you'll have to wail until Optical Flow support is introduced.

closed 2016-12-31 by digitalentity

liutairan, 2016-12-31:
@digitalentity So the slightly drift can be solved/corrected after I move my quadcopter to outdoor with GPS, right?

digitalentity, 2016-12-31:
Yes, GPS will allow the quad to compensate outdoors if you enable PosHold mode.

liutairan, 2016-12-31:
Great, I will try it tomorrow. Thanks~