iNavFlight/inav #1006: RX - Add support for non-inverted SBUS input

oleost, 2016-12-31:
Original Issue over at cleanflight : https://github.com/cleanflight/cleanflight/issues/1398

Original pullrequest over at cleanflight: https://github.com/cleanflight/cleanflight/pull/1604

Short description: For F3 target that can configure whether port is inverted or not, add abbility to configure it.

This is because apperntly some OpenLRS receivers uses non-inverted sbus signal.

digitalentity, 2017-01-11:
@oleost done!

digitalentity, 2017-01-28:
@oleost did you or someone else happen to test it?

oleost, 2017-01-28:
@digitalentity Never got an feedback from the original requester, tried PM him but no response.

However @DzikuVx said he was going to test it, unsure if he got the chance?

DzikuVx, 2017-01-28:
No. I did not had a chance yet

commit 889b14df (2015-06-14): Support pan and tilt on fixed wing.  Fixes #1006

The general idea is that nothing calls pwmWriteServo except for
writeServos() and that writeServos() knows which servos it has used.
PR #1080 (2017-01-18): SBUS inversion control (grafting from Cleanflight)
Fixes #1006
closed 2017-02-18 by digitalentity