iNavFlight/inav #1014: SERVO_TILT / Servo gimbal doesn't work on SPRacingF3 (release 1.5).

Xe54, 2017-01-01:
I can't get my gimbal working on SPRF3 acro board. Controls for pitch and roll axis are connected to the motor outputs 7 and 8 respectively. The way it is working with the current version of CleanFlight (1.14.2) is to enable SERVO_TILT in configurations and to set Servo 0 and Servo 1 to desired receiver channels (A2 and A3 in my case). Done. But it doesn't work that way in iNav. I would like to know if it's just a changed handling/channel mapping or is it a real bug? I also tried it with CHANNEL_FORWARDING and it doesn't work either.

Xe54, 2017-01-02:
Hi, I know this issue since iNav version 1.2.0, but it could be older. I could perform some more tests with older versions this weekend if necessary. My setup worked fine with each version of Cleanflight i've tried before (since 1.13.0). Here is my current dump:

# dump

# version
# INAV/SPRACINGF3 1.5.0 Dec 27 2016 / 10:12:30 (bfa1661)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature SERVO_TILT
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
[... 397 lines cut ...]
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 80
set rc_yaw_expo = 40
set thr_mid = 45
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

# 

Sorry, don't have bootlog at the moment, will post it at the weekend.

gurkenfolie, 2017-01-04:
inav 1.21 servo output works like it should, the newer tested versions didnt work for me with the same settings (tested 1.3 and 1.5).

dump:
# dump

# version
# INAV/SPRACINGF3 1.2.1 Sep 19 2016 / 08:24:09 (36954a8)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_NRF24
feature -SOFTSPI
feature RX_PPM
feature SERVO_TILT
feature GPS
feature FAILSAFE
feature ONESHOT125
feature BLACKBOX
feature CHANNEL_FORWARDING

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
[... 368 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 60
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 120
set pitch_rate = 100
set yaw_rate = 40
set tpa_rate = 0
set tpa_breakpoint = 1500

#

oleost, 2017-01-04:
Please provide dump of iNav 1.5 . Also remember you need to enable pwm output on latest iNav.

And also share "bootlog" dump from iNav 1.5

jflyper, 2017-01-06:
@Xe54 
- iNav 1.5.1 SPRACINGF3 Gimbal servo works as expected.
[image]

- You have different AUX channels assigned in your dumps (5 & 6 v.s. 4 & 5)

- Don't turn on channel forwarding. That does something totally different.

- bootlog dump is preferred.

Xe54, 2017-01-09:
OK, I spent a half of the weekent to figure out what the problem is. And I must confess, I still have no idea. Generally it seems to work, but not with my setup. 
My situation is:
1. My setup works properly with any version of Cleanflight
2. My setup doesn't work with iNav regardless of the version
3. A simplier iNav setup works with my 3D feiyu tech gimbal
4. Non of this iNav setups works with my 2D feiyu tech gimbal

I will need more time to figure out the reason. I know that my 2D gimbal is pretty sensitive to the input signals at start up. Could it be another behavior/timing of iNav compared to Cleanflight at start up, that constitutes a problem for my gimbal? Anyway here is the bootlog, I hope it helps somehow:

# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 501: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 501: 19   TIMER_CHANNEL_MAPPED  (7, 6, 1, 3)
 501: 19   TIMER_CHANNEL_MAPPED  (8, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (9, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (10, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (11, 6, 2, 3)
 501: 18  TIMER_CHANNEL_SKIPPED  (12, 6, 2, 3)
 501:  2          PWM_INIT_DONE 
 511:  3       EXTRA_BOOT_DELAY 
1048:  9         GYRO_DETECTION  (2, 0, 0, 0)
1163: 10          ACC_DETECTION  (3, 0, 0, 0)
1177: 11         BARO_DETECTION  (3, 0, 0, 0)
1178: 12          MAG_DETECTION  (2, 0, 0, 0)
2837: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2837:  4       SENSOR_INIT_DONE 
3388:  5          GPS_INIT_DONE 
3438:  7    TELEMETRY_INIT_DONE 
3489:  8           SYSTEM_READY 

#

oleost, 2017-01-11:
Okey, now I'm testing out `Servo Gimbal` feature

iNav servo tilt working on both pin 7 and pin 8
CLI: http://pastebin.com/V5raTg9x 
Bootlog: http://pastebin.com/GEn7LuvZ

I see my bootlog is different from yours.

oleost, 2017-01-12:
I started working on a Wiki page to assist in using Servo gimbal feature.

https://github.com/iNavFlight/inav/wiki/Connecting-a-Gimbal-to-iNav

Xe54, 2017-01-12:
Thanks, nice Wiki page. Can you please also explain, what live mode makes?

oleost, 2017-01-12:
Live mode is simple that the change you make in that tab is "live".

You dont need to press save to test example new midpoint, just enter new midpoint and it set right away instead of needing to press save.

Xe54, 2017-01-14:
Ok, I think I found something. As soon as I enable Serial RX for UART1 my gimbal stops to work. Can you check this please? I'm using UART1 for SBUS communication with FC.

oleost, 2017-01-14:
UART1? That's shared with USB?

Xe54, 2017-01-15:
Forget all I said above. I found the reason and that's not iNav. The problem was caused by BEC that I'm using. It seems to provide some interferences/noises over power supply, that are confusing my gimbal. For the reason I can't explain it works fine with CF. Anyway exchanging of BEC solved the problem. Here is the peace of hardware that I used as BEC
[image]
The bug is closed.

closed 2017-01-15 by Xe54

digitalentity, 2017-01-15:
@Xe54 great that you managed to sort it out! Thanks for letting us know!