iNavFlight/inav #1025: SPRacingF3 accelerometer calibration failed

dejz, 2017-01-02:
I can't get accelerometer calibrated on SPF3 acro board in newest release (1.5 stable) - Accelerometer calibration failed in Pre-arming checks. Acc hardware is recognized properly (MPU6050) and everything works okay in older releases (e.g. 1.4 stable).

Here is my current dump:

# dump

# version
# INAV/SPRACINGF3 1.5.0 Dec 31 2016 / 02:15:51 (37afad6)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature FAILSAFE
feature BLACKBOX

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
[... 388 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

DzikuVx, 2017-01-02:
Have you done calibration exactly like here: https://github.com/iNavFlight/inav/wiki/Sensor-calibration#accelerometer-calibration-steps  ?

dejz, 2017-01-02:
Yeah, I did not do it right. It helped. I run through the documentation and haven't noticed this procedure. I'm sure it will help if it will be better exposed.

DzikuVx, 2017-01-03:
In this case I'm closing this. Feel free to open if you find a bug

closed 2017-01-03 by DzikuVx