iNavFlight/inav #1029: Free mag alignment

Viald, 2017-01-03:
Is it possible to set any value to the mag alignment setting ?
The assembly of my quadcopter set the magnetometer to 135° CW but currently the only available values are 0, 90, 180 and 270°

[image]

oleost, 2017-01-03:
Atm there is no "proper way of doing this. But there is somehow possible

[image]

[image]

[image]

[image]

You could try and report back.

But seeing as your atleast the 3. guy asking for this it may be sensibile to allow setting a custom value?

So instead of fixed values in CLI, it would be freely adjustable, and then have the configurator only list the common ones as it currently is ?.

https://www.rcgroups.com/forums/showpost.php?p=36422701&postcount=9639

https://www.rcgroups.com/forums/showpost.php?p=36423155&postcount=9646

https://www.rcgroups.com/forums/showpost.php?p=36423422&postcount=9651

https://www.rcgroups.com/forums/showpost.php?p=36423488&postcount=9653

Viald, 2017-01-03:
Thanks for your feedback I'll give a try soon and report here.

andre92, 2017-01-04:
@oleost that will not work when the craft is tilted. Magnetic inclination can be as high as 60 degrees depending on where you live. There it will go wrong pretty quickly. The FC needs to know that the sensor is rotated 45 degrees. Proper solution would be free adjustment as OP asked, or maybe just add more options for 45 degrees. As long as zero settings are applied before compensating for alignment it should work. (calibrate the axes of the sensor, not the axes of the craft)

digitalentity, 2017-01-04:
Mag is read at a fairly low rate so more options transformation shouldn't tax us too high. We can also add an option to mount the mag at a 90deg vertical angle (i.e. if FC is mounted vertically and mag/GPS is horizontal)

Viald, 2017-01-05:
After calibration, the behavior of the mag is very very strange, sometime it's stable, something not, without trying to fly, just lay on the table...

Viald, 2017-01-05:
Would be nice :)

Viald, 2017-01-07:
Does it could be fixed ?

Viald, 2017-01-10:
Any chance to have a fix or a workaround ?

csurf, 2017-02-09:
What's the status on this? I urgently need a fix for a mag sensor that's mounted on a flying wing (popwing). The sensor is aligned parallel with the leading edge of the wing, while the FC is mounted "normally", with arrow pointing forward toward nose of plane. I need the ability to set an arbitrary angle for sensor X and Y axes.

digitalentity, 2017-02-09:
I'll tackle this once we have 1.6 out of the door

csurf, 2017-02-10:
how difficult would it be to implement this? How soon can a patch be developed? I'd be willing to compile & test a patch if you can come up with something quickly.

This will put a couple of my builds on hold, as the mag is at an angle on both wings.
Is there currently any issue with disabling mag on fixed-wing?

andre92, 2017-02-10:
@csurf Fixed wings can fly with gps only. Since fixed wing flies forward all the time, it can derive its course to a heading.

Buzzliteyear, 2017-07-09:
Just finished a build and needed a custom Mag alignment if this is possible, happy to do the testing

https://youtu.be/GXctoWFttoY?t=3m56s

giacomo892, 2017-07-09:
@Buzzliteyear you may want to share the code via a pull request :)

Buzzliteyear, 2017-07-09:
@giacomo892  I was brought up with Dos6.0 and find github 10 times harder to use but i shared in pull request hope its correct

giacomo892, 2017-07-10:
I don't see any pull request in the list. Just upload your code branch to your GitHub and then from the website do a pull request against development branch.

Buzzliteyear, 2017-07-10:
Ah that's because i dont know what to do ..........I wish so much your comments were easy to follow but its not sorry

giacomo892, 2017-07-10:
Just send the modifications and i'll do the pull request for you or otherwise read https://github.com/iNavFlight/inav/blob/master/docs/development/Development.md

Buzzliteyear, 2017-07-10:
heres the file at the end thanks
https://www.rcgroups.com/forums/showthread.php?2914402-BiCopter

giacomo892, 2017-07-10:
That's the compiled hex. We need the source code :)

Buzzliteyear, 2017-07-10:
I dont have it going that is why im asking how to request the feature

fiam, 2017-08-12:
For anyone else having this problem, here's a temporary workaround. The angle of rotation is hardcoded to 45º CW in the YAW axis, but should be pretty easy to modify the code to change your needs.

```diff
diff --git a/src/main/sensors/compass.c b/src/main/sensors/compass.c
index 5ef02d60..09e0673b 100644
--- a/src/main/sensors/compass.c
+++ b/src/main/sensors/compass.c
@@ -346,7 +346,13 @@ void compassUpdate(timeUs_t currentTimeUs)
         }
     }
 
-    alignSensors(mag.magADC, mag.dev.magAlign);
+#define SIN_M45 (-0.70710678118)
+#define COS_M45 (-(SIN_M45))
+    const int32_t x = mag.magADC[X];
+    const int32_t y = mag.magADC[Y];
+    mag.magADC[X] = x * COS_M45 - y * SIN_M45;
+    mag.magADC[Y] = x * SIN_M45 + y * COS_M45;
+    // alignSensors(mag.magADC, mag.dev.magAlign);
 
     magUpdatedAtLeastOnce = 1;
 }
```

fiam, 2017-08-19:
@digitalentity Since I'm now suffering this problem, I'm interested in fixing it so I don't have to keep using custom builds on my Mini Talon. Could we get some conversation going about how we should solve it?

I'd say these days the magnetometer is the only sensor very likely mounted outside the FC, so it's not worth to change the way we handle alignment of all sensors. Instead, I like @oleost's suggestion in https://github.com/iNavFlight/inav/issues/1029#issuecomment-270761469, which consists in introducing 3 new settings, one for each axis of rotation.

Buzzliteyear, 2017-08-20:
yes in 1 degree increments on the three axis is fine so it can be tuned once installed

commit 1a4a3e88 (2018-02-24): Allow aligning the mag to any rotation

Introduce 3 new variables which allow setting the decidegrees
for the mag sensor alignment. When any of these 3 variables
are non-zero, mag is assumed to be mounted off-board and
"align_mag" as well as the board alignment are ignored.

Settings are named align_mag_roll, align_mag_pitch and
align_mag_yaw.

Fixes #86
Fixes #1029
PR #2833 (2018-03-23): Allow aligning the mag to any rotation
Introduce 3 new variables which allow setting the decidegrees
for the mag sensor alignment. When any of these 3 variables
are non-zero, mag is assumed to be mounted off-board and
"align_mag" as well as the board alignment are ignored.

Settings are named align_mag_roll, align_mag_pitch and
align_mag_yaw.

Fixes #86
Fixes #1029
commit 948f2aed (2018-03-23): Allow aligning the mag to any rotation

Introduce 3 new variables which allow setting the decidegrees
for the mag sensor alignment. When any of these 3 variables
are non-zero, mag is assumed to be mounted off-board and
"align_mag" as well as the board alignment are ignored.

Settings are named align_mag_roll, align_mag_pitch and
align_mag_yaw.

Fixes #86
Fixes #1029
closed 2018-03-23 by fiam