iNavFlight/inav #1048: BLHeli interface Naze32

tonyintn, 2017-01-05:
This is not a bug just a question I am using the latest 1.51 and have made a custom flash file for my Naze32 by removing the blackbox and adding the LEDs. (Thanks again for the help on how to do this!).

I do under that reading thru the Issues that the BLHeli interface was removed from the Naze32 target due to memory of the F1. 
My question is since I have removed the blackbox is it possible for me to add it to my custom build if so which like would I need to change/add. I have looked at the common and naze32 target but unsure which one to add back into the build or if this is even possible.

Thanks in advance.

closed 2017-01-05 by tonyintn

handsomejackuk, 2017-01-11:
Hi Tony, wold you mind sharing this build with me as ihave original naze32 and wold love to get blheli interface working too... and i dont use blackbox either thanks...

HaseebNain, 2017-02-21:
@handsomejackuk Simply go into the target.h file and uncomment the line 

    #define USE_SERIAL_4WAY_BLHELI_INTERFACE 

it can be found here: [src/main/target/NAZE/target.h](https://github.com/iNavFlight/inav/blob/master/src/main/target/NAZE/target.h#L198)

and then add in the following after that line.

    #undef BLACKBOX
    #undef TELEMETRY

tonyintn, 2017-02-21:
Sorry for the delay, I will post a link that will enable Serial_4Way and LED here in a bit