iNavFlight/inav #1049: 1.6.0 (92d8ff86) omits CLI variables

stronnag, 2017-01-05:
1.6.0 (92d8ff86) omits a number of important CLI variables from CLI get, dump and diff commands, for example 
````
get acc
get mag
````
Does not return important variables (for backup and restore) such as acczero*, accgain* and magzero*, also many of the align* variables.

CLI is usuable, firmware is not testable.

stronnag, 2017-01-05:
git bisect tells me:
````
5e813d03da30add314f61ed3cd5aa79debde2aa6 is the first bad commit
commit 5e813d03da30add314f61ed3cd5aa79debde2aa6
Author: Martin Budden <mjbudden@gmail.com>
Date:   Sun Jan 1 17:56:26 2017 +0000

    Added gyroConfig parameter group

````
However, this may not actually be the real failing commit  as I flagged a number of prior commits in the bisect process as bad because:

* They didn't even compile (for NAZE); or
* They failed to boot 

This makes development useless for testing other new features (cli-diff, notch, msp_nav_poshold et al).

martinbudden, 2017-01-06:
@stronnag , a bit more info.

The CLI variables converted to parameter groups are here: https://github.com/iNavFlight/inav/blob/development/src/main/fc/serial_cli.c#L491-L560

The CLI variables not yet converted are here https://github.com/iNavFlight/inav/blob/development/src/main/fc/serial_cli.c#L571-L845

Over the next week or so the first list will get larger and the second list will get smaller. At some point we'll switch over to using the first list and from that point onwards the variables in the second list won't be available in the CLI. The point of switchover is a matter of judgement. Since you are doing a lot of testing I'm happy to switch over when you think it is beneficial.

stronnag, 2017-01-06:
Thanks for the update. I'll wait patiently :)

stronnag, 2017-01-07:
Closing, understand what's happening. I think this is all or nothing. We can really only switch when list 2 is zero entries.

closed 2017-01-07 by stronnag