iNavFlight/inav #1052: motor mixer for FLYING_WING and AIRPLANE only supports single motor setup

csurf, 2017-01-06:
I just tried configuring a flying wing using the second motor output on pin #2, but found that it isn't working. After taking a look at the code, I found that these two templates currently only support a single motor on output pin #1. The motor layout diagram within configurator for these two fixed-wing templates shows that both outputs #1 and #2 should be usable for either single or dual motor setups.

Here's the mixer code that shows both templates implementing the 'mixerSingleProp' motor template for single-motor setups:
https://github.com/iNavFlight/inav/blob/master/src/main/flight/mixer.c#L215
https://github.com/iNavFlight/inav/blob/master/src/main/flight/mixer.c#L235
https://github.com/iNavFlight/inav/blob/master/src/main/flight/mixer.c#L194

digitalentity, 2017-01-06:
I'd say we should fix this for 1.6 - simply making default mixer use the same mix both motor outputs is enough. It can be extended for differential thrust quite easily.

Schalonsus, 2017-01-13:
Just wanted to remind that there is an issue with differential thrust in pasthrough
#933

Linjieqiang, 2017-01-14:
If I use the same timer how can I devide the 50Hz for servos and 400Hz for motors in my fixwing?

digitalentity, 2017-01-14:
@Linjieqiang impossible, but in general on airplanes motors are fine with 50Hz update rate as well.

Linjieqiang, 2017-01-14:
Oh???Can I use 50Hz to update the motor?

digitalentity, 2017-01-14:
On airplanes - yes.

Linjieqiang, 2017-01-14:
OK.I try it and it shows well.

digitalentity, 2017-01-15:
Fixed in #1052

closed 2017-01-15 by digitalentity