iNavFlight/inav #1060: 1.5.1 - Board stops responding to MSP when GPS is used

digitalentity, 2017-01-07:
F4 based boar, connected to PC via MSP.

When GPS is not connected everything works perfectly. As soon a GPS is connected and starts sending out data board stops responding to MSP after a while. LEDs keep blinking so seemingly board is alive, although I'm not sure it is flyable.

digitalentity, 2017-01-07:
Correction - this is unrelated to MSP - board just freezes faster.

digitalentity, 2017-01-07:
Board even stops responding to SerialRX - can't ARM it. Looks like the serial system is messed up somehow.

digitalentity, 2017-01-07:
Hm... the bug is very sporadic - sometimes the board works fine, sometimes hangs up after a few seconds. I wonder what causes the bug.

jflyper, 2017-01-07:
Is this specific to F4?

digitalentity, 2017-01-07:
I have a strong feeling that this is a hardware issue. I've been flying this board today for half an hour without any issues. Also had it sitting connected to PC for quite a while without issue.

rb1205, 2017-01-09:
Is the GPS on a softserial? I had something like this happen when I had two soft serial ports working on MSP simultaneously (one to a MW_OSD and the other one to a Bluetooth serial interface connected to a PC running the GUI). After few seconds of polling statuses the board (SPRacingF3 evo) froze. Happened costantly every time.

tonyintn, 2017-01-09:
I also had this issue a couple of version back on a NAZE32 Rev6 Full version; it would happen if I sit idle once the GPS got a fix; sometime before if is just sits there it would just freeze and not respond I was afraid to fly it feeling that it would freeze while in the air but it seemed that once it was armed the freezing would not take place.   I two also think is has something to do with the combination of (GPS + MW_OSD). 
I was weird and I did not want to take chances so I replace it FC with a SPRacing

ABLomas, 2017-01-10:
Maybe it's related - SPRACINGF3 board in airplane. MSP and OSD (MW-OSD) ar on the same port, UART1. Works fine on USB-only (so GPS and other stuff is not powered on). Works fine on battery-only (so UART1 is used for OSD data) (both cases - ~25min on desk, not flying - but working fine))
Freezes in few seconds with USB+battery connected (MSP over USB + GPS connected). So far, 8 in 8 tries ;-)
Can't say if that happens in older versions - just swapped from CC3D controller and flashed new version.
Freezes completely - no inputs from RX, OSD is not updating, no servo movement/stabilisation.

rb1205, 2017-01-11:
MSP is not a bus, you can connect at most one running device at a time, and it's expected that stuff goes bad when you have more tha one device sending data to the same port at once, I don't really thing is related to the bug discussed.

wcalvert, 2017-01-30:
I believe I'm experiencing this issue. My hardware: http://www.readytoflyquads.com/flip-32-f4-omnibus-v2-pro Link to status, bootlog, and dump: https://paste.ee/p/oxbMt Link to build info: https://paste.ee/p/jsayv

DzikuVx, 2017-01-31:
My 2 F4 boards were running MSP and GPS yesterday for an hour each. Different UARTs.
No problems detected, I'm closing it since it does not seems to be software related

closed 2017-01-31 by DzikuVx