iNavFlight/inav #1067: INav and Hardware Failure Omnibus F3

UAV112, 2017-01-07:
I have had several Hardware failure notifications on the Omnibus F3. I am using GPS, SD Card as of right now.If i go back to Betaflight i do not have the flashing red light that indicates hardware failure.

digitalentity, 2017-01-08:
At first boot after flashing board is doing autodetection and non-detected sensors *may* be reported as failed. Try rebooting the board.

What sensor icons go red in the Configurator?

CLI `dump` and CLI `bootlog` will also help troubleshooting.

UAV112, 2017-01-08:
So, had to make sure all port where set i.e. SBUS, GPS, and do some calibrations i.e., accelerometer, GPS. the only way to get rid of the Multiple hardware errors was to unplug USB push the REBOOT button then replug. then unplug and then replug the USB so i could get connected. That seems to the ended all the hardware errors. Before they were happening every time i plugged in.

Thanks for your quick reply.

DzikuVx, 2017-01-24:
seems like problem solved, closing

closed 2017-01-24 by DzikuVx