iNavFlight/inav #1071: PPMSUM compatibility with UltimateLRS

PacciK, 2017-01-08:
Would be nice if the developer could open the PPM_SUM input parameters to accept the 16Channel PPMSUM stream as generated by UltimateLRS.

Here are some Oscilloscope grabs. Spave between each frame seems to be strictly 4.00ms irrespective of how many channels used, and at which position they are at ( even 2000uS or more )

Thanks a Million !

[image]
[image]
[image]

PacciK, 2017-01-09:
Felt the need to add that, on failsafe, an UltimateLRS receiver will output just 1 pulse ( the 16 channels disappear, and just the start frame remains ). If needed, I am very help to help with further screenshots, even a video clip to explain.

digitalentity, 2017-02-07:
@PacciK can you please test the attached hex file?
[inav_1.6.0_SPRACINGF3_ppm16.hex.gz](https://github.com/iNavFlight/inav/files/758118/inav_1.6.0_SPRACINGF3_ppm16.hex.gz)

wilco1967, 2017-02-07:
Hi Konstantin,

Where could I find a version for the SPRACINGF3EVO ?
I can compile myself, but where is it hidden ?

digitalentity, 2017-02-08:
@wilco1967 you can checkout the `ppm-16-channel` branch

wilco1967, 2017-02-08:
Hi Konstantin,

yep, that seems to work.... 

Just flew it a few meters behind my home (it's already dark outside) using the ULRS receiver on PPM, but seems to work fine....
Probably Saturday before I can attempt a proper flight furhter out.

I didn't check all 16 channels but 1-8 and also 16 work fine, so 9 to 15 are probably also fine.

Only 'issue' is they calibration is slightly off.... need to go to +512 (max) on output ranges in the taranis to get close to 2000 uS. on the receiver tab in iNav...  

With the Frsky receiver, I'm pretty close to +500 (+1500) to get 2000 on iNav (also PPM mode)
probably says something about the accuracy of the 433 modules I use... ;-)
On the TX side, I use a OrangeLRS receiver 100 mw module configured as TX... The receiver side is a Wolfbox receiver module (100 mw) 
Telemetry works fine at 19200 on UART1, 57600 on the bluetooth side...

Oh, and for the record.... the 'normal' 8 channel PPM from the Frsky, also still works fine

Now if I plan to keep flying on 433, I need to get the telemetry from the 433 module onto the Taranis screen.... There is some project doing just that using a Teensy... probably my next modification... ;-)

Thanks a lot !!!
I guess you will include this in the next release ?

closed 2017-02-09 by digitalentity

wilco1967, 2017-02-10:
Just for info.

I flew it out today using ULRS for control.... 
As far as iNav, all worked fine, including failsafe.

However, not so happy with the current range of my ULRS setup. Started to get problems with control when only 200 meters out... Both receiver and Tx are only 100 mw modules, and the receiver only has the stock Wolfbox antenna which probably is to blame, but I'll return to my trusted Frsky...

The Transmitter side is a 100 mW orange RX module (configured as a Tx, with a 'proper' Goya NA775 antenna), the receiver side is a wolfbox 100 mW rx with the stock antenna)

I didn't get the ULRS for control, but only for telemetry... just wanted to see how it worked, and I know that now. ;-)
I never fly out further than say 1 km (also range limit for my video link), and Frsky will do that without problems.
Thanks anyway for the 16 Channel PPM... I'm sure there are many other people thankful for that !

digitalentity, 2017-02-11:
@wilco1967 433MHz is tricky. Good antennas matter a lot - I advise you replace stock antenna ASAP - Chinese often mix then up - it might be 5.8G or 2.4G instead of proper 433MHz. IRC Sander-style antennas work quite good.

Also, cheap receivers lack low-pass filter on the antenna which significantly reduce performance. I don't know if WolfBox has it or not.

For LRS I'm using TBS CrossFire with great success - I'm more likely to lose the video feed rather than control.

wilco1967, 2017-02-11:
yeah, I know.....
But the stock antenna 'looked' like a proper antenna, very much like a sander style... definitely not a 5.8 / 2.4
From what I've read, the wolfbox doesn't have the filter... Previously, I had also an orangeRx, which had a much better range (but that one was lost on the fly-away ;-).

The Crossfire is a little above my budget I'm afraid ;-)....