iNavFlight/inav #1072: How about add a new feature to enable auto level trim?

liutairan, 2017-01-09:
I am considering is it possible to have a new feature to enable auto level trim. When the accelerometer is not properly trimmed, the quadcopter may have drift to one direction. Then the accelerometer will have constant value reading in some direction. Then it can use the readings from accelerometer to make corrections so that the quadcopter not drifting any more.

I am actually suffering from the drifting problem. Every time when my quadcopter hit the ground, the board will have small angle changes. Then I need to retune the board and sensor alignment again, and it seems like I never able to have a real POSHOLD and ALTHOLD. I think if it can use the knowledge of accelerometer to adjust the drifting problem, the POSHOLD will be much better.

How do you think about this?
Thanks~

DzikuVx, 2017-01-09:
https://github.com/iNavFlight/inav-configurator/issues/61 ?

liutairan, 2017-01-09:
@DzikuVx I think it is pretty similar. I think that one actually means when FC is still on the ground? If we can adjust that in the process of flight, that will be great. Because once the quadcopter taking-off, it may hit the ground of any object if manual control it, then the flight will be awful after that.

DzikuVx, 2017-01-09:
In the process of flight you can use https://github.com/iNavFlight/inav/blob/master/docs/Inflight%20Adjustments.md

There is no way to do fully automatic inflight acc level calibration since FC does not have "secondary" sensor that says when UAV is level. GPS can not be used, since it also takes wind drift into account

liutairan, 2017-01-09:
@DzikuVx Thanks for that page. 
I think when UAV is not level and drifting there will be readings from acc sensor, so this reading from acc sensor is not enough to indicate it is not level when there is no wind? Yes you are right about the wind drift. This is really a problem.

rb1205, 2017-01-11:
You're missing the point. The drift is caused by a misalignation between the accelerometer and the powertrain orientations. The FC already uses the accelerometer reading as an error feedback to level the mc, so the (average) ouput of the accelerometer on the x and y axes are (or should be) zero even while it's drifting, 

In other words, in order to perform an auto-calibration the FC needs an external frame of reference to know when it's "still". It may work using the GPS, but i don't know if it's precise enough.

digitalentity, 2017-01-11:
@liutairan Actually @rb1205 is correct. As far as copter is concerned it is flying perfectly level, however if thrust vector is not aligned with FC Z-axis aircraft will be drifting. FC is unaware of such misalignment. Theoretically GPS can use used to compensate, but it's complicated, not reliable and will probably introduce more bugs than fix.

liutairan, 2017-01-11:
@rb1205 Yeah, I see, you are right. This is really a problem. I think my drift problem actually comes from the hard landing or collision with the ground so that the thrust vector keeps change every time it hits the ground then the board alignment changes.

liutairan, 2017-01-11:
@digitalentity You are right, that is the real reason of the drifting. I think my drift problem actually comes from the hard landing or collision with the ground so that the thrust vector keeps change every time it hits the ground then the board alignment changes. I really should change to some other frames which the quadcopter arms and body connection more stable but unlucky I am not able to. Thanks~

liutairan, 2017-01-11:
@digitalentity Then is it possible to change the acc alignment angles during flight? For example, can I observe the UAV still or not, then use some settings to change the board alignment values and save it rather than using the joystick to adjust them? Because I wish to have autonomous flight without joystick only with a ground control station and I have a swarm of quadcopters to do the autonomous flight I really don't have so many hands to operate them. LOL.

rb1205, 2017-01-12:
There is already, it's called inflight accelerometer calibration. But still, even if you calibrate your accelerometers perfectly for a given day you'll still have drift: , ground/wall distance, CoG position, wear, temperature... every slight difference in any of those is going to produce drift to some degree, and that's not even considering wind. To automatically prevent trim you need a GPS, which is what this fork is about BTW.

liutairan, 2017-01-13:
@rb1205 Actually I do have a GPS, I thought if we can make it work better without GPS, then it will do more than better with GPS. Because GPS might not very accurate for small distance errors, I guess adding GPS will make it work better but still have some limitations. OK, I see now. Thanks you~

By the way, the inflight accelerometer calibration http://tldrify.com/elw given by https://github.com/iNavFlight/inav/wiki/Multirotor-guide said "... attempted to calibrate with this procedure and I was not successful ..."

closed 2017-01-28 by liutairan