iNavFlight/inav #1090: Sonar - refactor the code to use detection and configuration like all other sensors

digitalentity, 2017-01-13:
Create `rangefilder_hardware` parameter and make it control what hardware is used.
Remove `SONAR` feature.

hali9, 2017-02-04:
How to enable sonar on SPRACINGF3 on inav 1.6 ?
I set in cli:
_set rangefinder_hardware=HCSR04_
but after reboot I get:
_# get rangefinder_hardware
rangefinder_hardware = NONE
Allowed values: NONE, HCSR04, SRF10_

Is any other parameter or setting or maybe pin change?
Sonar work in 1.5.1 and I no hardware change.

My bootlog:
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 502: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (9, 9, 0, 2)
 503: 19   TIMER_CHANNEL_MAPPED  (10, 10, 0, 2)
 503: 19   TIMER_CHANNEL_MAPPED  (11, 11, 0, 2)
 503: 19   TIMER_CHANNEL_MAPPED  (12, 12, 0, 2)
 503:  2          PWM_INIT_DONE 
 549:  9         GYRO_DETECTION  (2, 0, 0, 0)
 665: 10          ACC_DETECTION  (3, 0, 0, 0)
 686: 11         BARO_DETECTION  (2, 0, 0, 0)
 686: 12          MAG_DETECTION  (2, 0, 0, 0)
2350: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2350:  4       SENSOR_INIT_DONE 
2852:  5          GPS_INIT_DONE 
2903:  8           SYSTEM_READY

DzikuVx, 2017-02-04:
I don't think this is finished yet

DzikuVx, 2017-02-04:
Or rather: SONAR support is disabled in INAV 1.6

closed 2017-03-02 by digitalentity