iNavFlight/inav #1110: ROVER support?

csurf, 2017-01-17:
not urgent, but are there any long-term plans to implement some kind of ground vehicle/rover support (cars, boats, subs, etc.) to iNav?

DzikuVx, 2017-01-17:
There were some discussions on that topic. Should be doable :)

csurf, 2017-01-20:
One thing I'll leave as a suggestion is to support reverse throttle & mechanical braking (versus simply using 'active motor braking'). I don't think even APM rover supports this...

DzikuVx, 2017-01-20:
First we would have make IMU unaware of altitude and make it working nicely only in 2D

digitalentity, 2017-01-20:
@DzikuVx it does already - we can just ignore altitude if necessary. I have a feeling that ROVER mixer will be very similar to airplane.

csurf, 2017-01-20:
for a very simple implementation, pretty much all IMU math is unecessary. Only 2D position & speed from GPS are necessary. Steering & speed based on GPS data would allow for simple nav functions (waypoints, RTH, pos hold, etc.)

for more a more advanced setup, perhaps IMU math could possibly be used for stuff like:
- safety measures to prevent vehicle from climbing/descending/traversing steep slopes & flipping over
- maybe some crazy feature to auto-flip (via mechanical arm?) a vechicle that rolled over
- limit vehicle velocity over rough surfaces (bumpy roads/rocks) to prevent damage, crashes, or shaky video (based on vibrations/IMU readings).
- increase power/throttle on steep climbs, and apply brake on steep descents (based on IMU angles)

any other ideas???

biolippi, 2017-03-14:
Would it actually already be possible with the current version? Just connect motor and rudder?

digitalentity, 2017-03-14:
@biolippi you are free to try. Rover is a bit different from airplane - no altitude control, no banking control, PIDs attenuated by throttle, no speed control etc.

michaelvanderhoek, 2017-09-26:
would love to see support for boats. I'm thinking of a simple setup for my baitboat!
making some waypoints and fish on the same spot every time. 

Maybe an possibility to save the waypoints even when the power is off?

jamesnabbott, 2018-09-04:
[Just saw this video](https://www.youtube.com/watch?v=tkAWrsIbuh8) where it looks like someone was able to setup iNav on a rover. I'm not sure what level of code modification he's done but I have an ArduRover AP Rover setup already and would love to try and get it going with iNav.

paralin, 2019-10-16:
Working on this.

closed 2020-04-23 by DzikuVx