iNavFlight/inav #1111: add safeguard to serial processing for 'R' char / bootloader mode entry

csurf, 2017-01-17:
Some bluetooth modules output verbose connection info in 'AT' command format...
example on manual disconnect:
`+DISC=SUCCESS
+READY
+PAIRABLE`

The 'R' in the above '+READY' output message will cause the board to go into bootloader mode. Per @stronnag's suggestion, perhaps there should be a check/timeout to make sure that only when a single 'R' char is sent will the firmware go into bootloader mode (i.e., no further characters are received after the R).

digitalentity, 2017-01-17:
This is tricky. The thing is that we don't have any place to schedule the delay for rebooting into bootloader - the reboot happens instantly after we receive an 'R' character.

martinbudden, 2017-01-17:
Why is an 'R' character used, why not something more obscure like '~' or '$' ?

stronnag, 2017-01-17:
Agreed, but perhaps require a quiet period prior to the R. By recording millis for the last received character. Even 500ms would fix the buggy client issue and be easily done in make and the UI

stronnag, 2017-01-17:
Isn't $ the lead on for MSP. And the character in configurable in the CLI.

digitalentity, 2017-01-17:
We can probably add a dummy `delay()` for i.e. 100ms and check if there's any more characters in the serial receive buffer. If there is - don't reboot. This will cause a minor 100ms hickup in processing the data but will definitely solve the garbage issue.

digitalentity, 2017-01-17:
I wouldn't change the default reboot character as it will break Configurator compatibility with previous versions.

csurf, 2017-01-17:
side-note, but it would seem like a good idea to be able to manually change the reboot character in configurator as well. Or, perhaps have configurator query firmware and retrieve the configured reboot char?

digitalentity, 2017-01-17:
@csurf can you test the #1113?

csurf, 2017-01-17:
ok, will test later today/tonight & post back...

csurf, 2017-01-18:
just built & tested.
Now, there seems to be a problem with the CLI. The exit/save commands aren't correctly rebooting when in CLI, even when connected via USB. 

Plus, when I run 'dump', there's a quick 'beep' at about the halfway point during the scrolling output. 
Also, after trying to save/exit, the EEPROM is being corrupted (all settings are set back to default).

I don't know if these issues are related to the 1.6 dev version, the version of gcc I'm using (5.4q2), or if the serial delay patch is simply breaking everything.

Let me know if you need more info...

stronnag, 2017-01-18:
Happens regardless of this patch on development  at the moment. Possibly a result of pg changes?

csurf, 2017-01-18:
@digitalentity, can you help me out by getting me a patched naze build of the release version? I suck at git...

DzikuVx, 2017-01-18:
I'm on it

DzikuVx, 2017-01-18:
@csurf https://drive.google.com/file/d/0Bx6X245zlrsnS2E1bks4ZFNZeDA/view?usp=sharing

csurf, 2017-01-18:
@DzikuVx isn't that a 1.6 build? I need a patched 1.5.1 build

DzikuVx, 2017-01-18:
ah... damn, my bad :) That was before my morning coffee :) Will make a new one

DzikuVx, 2017-01-18:
@csurf now it should be fine https://drive.google.com/file/d/0Bx6X245zlrsneDYzWUdoOTJvNUk/view?usp=sharing

csurf, 2017-01-18:
@DzikuVx 
thanks man, that worked.

@digitalentity, patch seems to work fine, no more issues with board going into bootloader on serial disconnect... looks good...

digitalentity, 2017-01-18:
Ok, merging and closing.

closed 2017-01-18 by digitalentity