iNavFlight/inav #1133: NAZA E300 esc beeping from INAV 1.4 on

damsport11, 2017-01-19:
Hello,
I have been trying to revive a F450 using the stock NAZA E300 ESCs, however, when I flash the newest INAV firmware (1.5.1) and connect the LiPo, the esc's start and continue beeping (beep...beep...beep). By accident I uploaded a old version of INAV and this works fine (standard NAZA riddle and motors are functional). I went back to older versions and the beeping starts from INAV 1.4 on. Has been tested on a Naze32 rev6 and SPracing F3 evo. Any idea how to get the ESCs working on the newest version?
Cheers.

digitalentity, 2017-01-19:
Did you enable PWM_OUTPUTS in Configurator?

damsport11, 2017-01-19:
...of course not... sorry bout that...
Thanks for the fast help and making great software ;-)

digitalentity, 2017-01-19:
No worries. Happy flying!

closed 2017-01-19 by digitalentity